mtu-most/franklin #1: Error during "make"

clemsept, 2016-02-13:
Hello, I've an error during the execution of 'make' on Debian Jessie and Raspbian
```
dpkg: dependency problems prevent configuration of franklin:
 franklin depends on python3-websocketd; however:
  Package python3-websocketd is not installed.
 franklin depends on python3-network; however:
  Package python3-network is not installed.
 franklin depends on python3-serial; however:
  Package python3-serial is not installed.
 franklin depends on python3-fhs; however:
  Package python3-fhs is not installed.
 franklin depends on python3-gi; however:
  Package python3-gi is not installed.
 franklin depends on libgtk-3-0; however:
  Package libgtk-3-0 is not installed.

dpkg: error processing package franklin (--install):
 dependency problems - leaving unconfigured
Processing triggers for man-db (2.7.0.2-5) ...
Processing triggers for mime-support (3.58) ...
Processing triggers for systemd (215-17+deb8u3) ...
Errors were encountered while processing:
 franklin
Makefile:21: recipe for target 'install-all' failed
make: *** [install-all] Error 1
```
All required packages have been installed.
Have you any ideas about this ?

clemsept, 2016-02-14:
I managed to manually install the packages. I have extracted the files in 'franklin/install/franklin-install.zip' and run `dpkg --install <package_name>`. `make` has correctly installed franklin.
However, the server does not respond in the browser after launched `franklin

wijnen, 2016-02-15:
Thanks for the report.

You shouldn't use the packages from the zip; they are out of date, and I will remove that file.

There was a bug in the Makefile which is now fixed.  I also made some changes which means you now need other packages then before.  The error message you saw mentioned them (in addition to the things Make failed to install).  If you pull the latest source and run make, after the error you should run ``apt-get install -f`` and that will install anything you need.  Because the zipfile contained outdated packages, you should also ``apt-get update; apt-get upgrade`` to get new versions for them.

I hope that fixes it.  If it still fails on make, please reopen this and if there's another problem, open a new issue.

clemsept, 2016-02-16:
Thank you so much.
I've removed every obsolete packages, delete and (re)clone the repository.
```
clement@debian:~$ cd franklin
clement@debian:~/franklin$ make
git submodule add https://github.com/wijnen/python-fhs || git submodule update python-fhs
The following path is ignored by one of your .gitignore files:
python-fhs
Use -f if you really want to add it.
error: pathspec 'python-fhs' did not match any file(s) known to git.
Did you forget to 'git add'?
Makefile:88: recipe for target 'module-python-fhs' failed
make: *** [module-python-fhs] Error 1
clement@debian:~/franklin$ sudo apt-get install -f
[sudo] password for clement:
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```
But there is a problem with .gitignore file.
I tried to remove lines about packages (I don't know if it's a good solution).
Finally, it's the same problem, the server is installed but do not respond.

wijnen, 2016-02-17:
Thanks again, it seems I broke the .gitignore indeed; I fixed it now.

Please try running Franklin from the commandline to see if it produces any messages.  For this, first do:
``sudo service franklin stop``
Then go to the location where you downloaded Franklin and go into the ``server`` directory.  From there, run
``DEBUG=2 ./server.py``

It should tell you about any attempts to connect to it, but perhaps it fails to start completely.  Let me know what the output is.

wijnen, 2016-02-17:
My previous comment is wrong; when running from the commandline, you need to pass the --cdriver argument.  More details are in the installation page on the wiki, which I just updated.  If you're running on a Raspberry Pi, I also know what was wrong and the solution is now documented there as well.

clemsept, 2016-02-17:
`Make` is working !
After the execution of `--cdriver cdriver/cdriver`,
I have these log :
```
clement@debian:~/franklin/server$ ./server.py --cdriver cdriver/vg
Wed Feb 17 21:44:56 2016 CET +0100: Franklin server is running
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gi/overrides/GLib.py", line 702, in <lambda>
    func_fdtransform = lambda _, cond, *data: callback(channel, cond, *data)
  File "/usr/local/lib/python3.4/dist-packages/network.py", line 402, in <lambda>
    self._event = GLib.io_add_watch(self.socket.fileno(), GLib.IO_IN | GLib.IO_PRI | GLib.IO_HUP | GLib.IO_ERR, lambda fd, cond: (self._line_cb() or True))
  File "/usr/local/lib/python3.4/dist-packages/network.py", line 405, in _line_cb
    self._linebuffer += self.recv(self.maxsize)
AttributeError: 'Socket' object has no attribute 'maxsize'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gi/overrides/GLib.py", line 702, in <lambda>
    func_fdtransform = lambda _, cond, *data: callback(channel, cond, *data)
  File "/usr/local/lib/python3.4/dist-packages/network.py", line 402, in <lambda>
    self._event = GLib.io_add_watch(self.socket.fileno(), GLib.IO_IN | GLib.IO_PRI | GLib.IO_HUP | GLib.IO_ERR, lambda fd, cond: (self._line_cb() or True))
  File "/usr/local/lib/python3.4/dist-packages/network.py", line 405, in _line_cb
    self._linebuffer += self.recv(self.maxsize)
AttributeError: 'Socket' object has no attribute 'maxsize'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gi/overrides/GLib.py", line 702, in <lambda>
    func_fdtransform = lambda _, cond, *data: callback(channel, cond, *data)
  File "/usr/local/lib/python3.4/dist-packages/network.py", line 402, in <lambda>
    self._event = GLib.io_add_watch(self.socket.fileno(), GLib.IO_IN | GLib.IO_PRI | GLib.IO_HUP | GLib.IO_ERR, lambda fd, cond: (self._line_cb() or True))
  File "/usr/local/lib/python3.4/dist-packages/network.py", line 405, in _line_cb
    self._linebuffer += self.recv(self.maxsize)
AttributeError: 'Socket' object has no attribute 'maxsize'
```
The connection turn around and do not respond.

wijnen, 2016-02-17:
Thanks.  That means the submodules are not up to date, and that is my fault.  I just changed the Makefile to make sure it will always pull in the latest versions of everything and rebuild with them.  Can you please pull the latest version of Franklin and run make again?

clemsept, 2016-02-17:
Thank's to you ;).
After clean, pull and make , it's the same error : `AttributeError: 'Socket' object has no attribute 'maxsize'`.

wijnen, 2016-02-17:
Ah, I see you have versions of the packages installed in /usr/local/.  That probably came from an earlier install attempt.  Remove them and it should work.

clemsept, 2016-02-18:
make failed after cleaning and removing everything :
```
[...]
dpkg-checkbuilddeps: Unmet build dependencies: debhelper (>= 9)
dpkg-buildpackage: warning: build dependencies/conflicts unsatisfied; aborting
dpkg-buildpackage: warning: (Use -d flag to override.)
debuild: fatal error at line 1376:
dpkg-buildpackage -rfakeroot -D -us -uc failed
+ test '' -a '' -a ''
+ exit 0
sudo dpkg -i 
dpkg: error: --install needs at least one package archive file argument

Type dpkg --help for help about installing and deinstalling packages [*];
Use 'apt' or 'aptitude' for user-friendly package management;
Type dpkg -Dhelp for a list of dpkg debug flag values;
Type dpkg --force-help for a list of forcing options;
Type dpkg-deb --help for help about manipulating *.deb files;

Options marked [*] produce a lot of output - pipe it through 'less' or 'more' !
Makefile:21: recipe for target 'install-all' failed
make: *** [install-all] Error 2
```
:S

wijnen, 2016-02-18:
> dpkg-checkbuilddeps: Unmet build dependencies: debhelper (>= 9)

You need to install debhelper (at least version 9).

clemsept, 2016-02-18:
Oops ! I forgot this package !
...
I purged everything in `/usr/local/lib` and run `make `: It WORKS !
Thank you very much ;)

closed 2016-02-18 by clemsept

ldurous, 2017-01-11:
Hello Bas and clemsept,

And thank you also Bas for your work, I'm trying to set up a syringe pump for my lab :)

I have actually the same problem. I tried to clone and install franklin on my raspberry pi several times but it says each time that your dependencies are not installed, even after i cloned them from your github page and run the command make..... If you could help me I would be very grateful !

Here is the code after running `make` :
`git pull
Already up-to-date.
git submodule foreach git pull
Entering 'python-fhs'
Already up-to-date.
Entering 'python-network'
Already up-to-date.
Entering 'python-websocketd'
Already up-to-date.
git submodule foreach ../mkdeb 
Entering 'python-fhs'
+ '[' '!' -d debian ']'
++ find . -name '.#*'
+ cvs_conflict_files=
+ '[' '' ']'
+ dir=/tmp
++ dpkg-parsechangelog
+ changelog='Source: python-fhs
Version: 0.2-1
Distribution: UNRELEASED
Urgency: medium
Maintainer: Bas Wijnen <wijnen@debian.org>
Date: Mon, 30 Nov 2015 11:24:26 -0500
Changes:
 python-fhs (0.2-1) UNRELEASED; urgency=medium
 .
   * New release.'
++ echo 'Source: python-fhs
Version: 0.2-1
Distribution: UNRELEASED
Urgency: medium
Maintainer: Bas Wijnen <wijnen@debian.org>
Date: Mon, 30 Nov 2015 11:24:26 -0500
Changes:
 python-fhs (0.2-1) UNRELEASED; urgency=medium
 .
++ grep '^Source: '
   * New release.'
++ cut -b9-
+ name=python-fhs
++ echo 'Source: python-fhs
Version: 0.2-1
Distribution: UNRELEASED
Urgency: medium
Maintainer: Bas Wijnen <wijnen@debian.org>
Date: Mon, 30 Nov 2015 11:24:26 -0500
Changes:
 python-fhs (0.2-1) UNRELEASED; urgency=medium
 .
++ grep '^Version: '
   * New release.'
++ cut -b10-
+ fullversion=0.2-1
+ version=0.2
+ debversion=1
+ echo name=python-fhs, version=0.2
name=python-fhs, version=0.2
+ rm -rf /tmp/python-fhs-0.2.orig
+ mkdir -p /tmp/python-fhs-0.2.orig
+ tar czf - .
+ tar -C /tmp/python-fhs-0.2.orig -xzf -
+ find /tmp/python-fhs-0.2.orig -name CVS -print0
+ xargs -0 rm -rf
[... 691 lines cut ...]
+ test '' -a '' -a ''
+ exit 0
sudo dpkg -i /tmp/franklin_0.2-1_armhf.deb
(Reading database ... 130424 files and directories currently installed.)
Preparing to unpack /tmp/franklin_0.2-1_armhf.deb ...
Unpacking franklin (0.2-1) over (0.2-1) ...
dpkg: dependency problems prevent configuration of franklin:
 franklin depends on python3-websocketd; however:
  Package python3-websocketd is not installed.
 franklin depends on python3-network; however:
  Package python3-network is not installed.
 franklin depends on python3-fhs; however:
  Package python3-fhs is not installed.

dpkg: error processing package franklin (--install):
 dependency problems - leaving unconfigured
Processing triggers for systemd (215-17+deb8u5) ...
Processing triggers for man-db (2.7.0.2-5) ...
Processing triggers for mime-support (3.58) ...
Errors were encountered while processing:
 franklin
Makefile:21: recipe for target 'install-all' failed
make: *** [install-all] Error 1

`

wijnen, 2017-01-12:
You are missing some packages.  I've added some checks to the build system, so it now gives an error message explaining what you're missing.  Please run "make clean", then make again and it should tell you what you need to install for the build to work.

ldurous, 2017-01-12:
Thank you a lot ! I installed doxygen, graphvidz and doxypy and `make` command ran without error. 
Now it seems to work, i can join the interface :)