mtu-most/franklin #3: Use compiled code for parsing g-code

wijnen, 2016-02-17:
Parsing g-code is very slow.  This can be improved a lot if it is moved from the python driver to the cdriver.

dktrived, 2017-05-29:
Hi, Could it be possible that a gcode is not loaded completely but it still shows up in the white box?

It was taking like 45 mins for parsing gcode so i refreshed the page and the code was there. When i printed it , it didnt print the whole thing . Was wondering if there would be any way to check that ?

I tried to load the code again, but again it took a very long time to parse and didnt finish. Not sure why this is happening.

wijnen, 2017-05-29:
No, this is not possible.  As long as Franklin is parsing the G-Code, it is completely unresponsive (temperature updates are also stalled), so it will not even send the machine's settings, and certainly not the list of jobs.

If it doesn't finish, it is possible that the driver crashed.  In that case, please send me the G-Code and your machine settings (the ini file) so I can try to reproduce it.

wijnen, 2017-12-29:
The G-Code parsing has been moved to cdriver in the latest commit.

closed 2017-12-29 by wijnen