mtu-most/franklin #5: Can't flashing Melzi board V2.0.

clemsept, 2016-02-18:
Hello, it's me... (Bad joke, I know ^^).
The franklin server is running and `@/dev/ttyUSB0` is detected.
I have a timeout with autodetect and when I want to flash, this pop-up appears :
```
error: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/websocketd.py", line 479, in _recv
    self._call(data[1][0], data[1][1], data[1][2], data[1][3])
  File "/usr/lib/python3/dist-packages/websocketd.py", line 496, in _call
    ret.send(None)
  File "/usr/bin/franklin", line 390, in upload
    assert ports[port] is None
AssertionError
```

wijnen, 2016-02-18:
That error (assert ports[port] is None) means that it is still trying to detect a printer, and therefore it cannot flash the firmware yet.  This is not a bug, but there is certainly room for improvement.  I'll leave this open as a feature request issue.  First a workaround: disable autodetect, or wait for autodetection to time out (this doesn't always happen though, so disabling autodetect is safer).

How I want it to behave:

- [ ] Show indicator in the interface that detection on the port is in progress.
- [ ] Add "abort detection and flash firmware" as an option, only visible while detection is in progress.

clemsept, 2016-02-18:
I also get the message : 
[image]

wijnen, 2016-02-20:
Ah, that is interesting; I remember I had trouble flashing a sanguinololu as well. Can you flash anything to it from the Arduino interface or using avrdude from the commandline?  If so, what setting do you use for the protocol (the -c argument of avrdude; if you enable verbose output in the Arduino IDE it will tell you the command it uses)?

clemsept, 2016-02-21:
From Arduino IDE, I can flash the melzi V2 with Sanguino driver, but results are very bad.
I never used avrdude, I will try it.

wijnen, 2016-02-21:
It seems I used the wrong protocol for standard Melzis; I never noticed because I always flashed optiboot into them.
I have now changed the server so the list of options is read from Arduino's files.  So if  you make sure arduino-mighty-1284p is installed when you build (and run) Franklin, you should be able to choose the "original" version.  Please let me know if that works.

clemsept, 2016-02-22:
Now, flashing options does not appears.
[image]

wijnen, 2016-02-23:
Do you have arduino-core and arduino-mighty-1284p installed?  There should be files named /usr/share/arduino/hardware/*/boards.txt

Does it complain about anything on the commandline?

clemsept, 2016-02-23:
Arduino-core and arduino-migthy-1284p are installed.
During flash, it's the same problem : `Upload done : null`.

On windows, in Arduino 1.0.6, it's work with Sanguino drivers :
I use it : [Sanguino Driver](https://github.com/reprappro/Marlin/tree/master/sanguino).

wijnen, 2016-02-23:
I'm sorry it's taking so long to find out what the problem is.  Please answer these questions, that should help me find the problem:
* Does /usr/share/arduino/hardware/mighty-1284p/boards.txt exist on the machine that runs the server?
* How can you flash if there are no buttons?
* Can you copy the messages you see in the commandline?

clemsept, 2016-02-24:
Sorry, I'm not clear in my explanations 

- `/usr/share/arduino/hardware/mighty-1284p/boards.txt` exist.
- I can see flashing options when I launch `./server.py` only. (It doesn't work when I launch the service).
- Log when I'm trying to flash melzi :
```
Wed Feb 24 20:51:02 2016 CET +0100: Franklin server is running
Wed Feb 24 20:51:07 2016 CET +0100: Error reading from socket: [SSL: TLSV1_ALERT_UNKNOWN_CA] tlsv1 alert unknown ca (_ssl.c:1769)
Wed Feb 24 20:51:07 2016 CET +0100: Error reading from socket: [SSL: TLSV1_ALERT_UNKNOWN_CA] tlsv1 alert unknown ca (_ssl.c:1769)
Wed Feb 24 20:56:05 2016 CET +0100: removing port /dev/ttyUSB0
Wed Feb 24 20:56:12 2016 CET +0100: detecting printer on /dev/ttyUSB0
Wed Feb 24 20:56:19 2016 CET +0100: skip non-id: 1ef870187f067c069ee07fe6 (b'\x1e\xf8p\x18\x7f\x06|\x06\x9e\xe0\x7f\xe6')
Wed Feb 24 20:56:20 2016 CET +0100: skip non-id: 001880e600180018e6009800 (b'\x00\x18\x80\xe6\x00\x18\x00\x18\xe6\x00\x98\x00')
Wed Feb 24 20:56:20 2016 CET +0100: skip non-id: 1873e600180018e600180018 (b'\x18s\xe6\x00\x18\x00\x18\xe6\x00\x18\x00\x18')
Wed Feb 24 20:56:21 2016 CET +0100: skip non-id: e6e60018001873e67c1e7fe6 (b'\xe6\xe6\x00\x18\x00\x18s\xe6|\x1e\x7f\xe6')
Wed Feb 24 20:56:21 2016 CET +0100: skip non-id: 7cf81ee07c1818e0181e70c6 (b'|\xf8\x1e\xe0|\x18\x18\xe0\x18\x1ep\xc6')
Wed Feb 24 20:56:21 2016 CET +0100: skip non-id: 80fe1e1e1ee6801800e60098 (b'\x80\xfe\x1e\x1e\x1e\xe6\x80\x18\x00\xe6\x00\x98')
Wed Feb 24 20:56:21 2016 CET +0100: skip non-id: 001873e67c1e7fe67cf81ee0 (b'\x00\x18s\xe6|\x1e\x7f\xe6|\xf8\x1e\xe0')
Wed Feb 24 20:56:21 2016 CET +0100: skip non-id: 7c1818e0181e70c680fe1e1e (b'|\x18\x18\xe0\x18\x1ep\xc6\x80\xfe\x1e\x1e')
Wed Feb 24 20:56:23 2016 CET +0100: skip non-id: 1ee6001800e600180018e600 (b'\x1e\xe6\x00\x18\x00\xe6\x00\x18\x00\x18\xe6\x00')
Wed Feb 24 20:56:23 2016 CET +0100: skip non-id: 180018e600180018e6009800 (b'\x18\x00\x18\xe6\x00\x18\x00\x18\xe6\x00\x98\x00')
Wed Feb 24 20:56:26 2016 CET +0100: skip non-id: 1873e600180018e600980018 (b'\x18s\xe6\x00\x18\x00\x18\xe6\x00\x98\x00\x18')
Wed Feb 24 20:56:26 2016 CET +0100: skip non-id: e600180018e600180018e600 (b'\xe6\x00\x18\x00\x18\xe6\x00\x18\x00\x18\xe6\x00')
Wed Feb 24 20:56:26 2016 CET +0100: skip non-id: 18001873e67c1e7fe67cf81e (b'\x18\x00\x18s\xe6|\x1e\x7f\xe6|\xf8\x1e')
Wed Feb 24 20:56:27 2016 CET +0100: skip non-id: e07c1818e0181ee600180018 (b'\xe0|\x18\x18\xe0\x18\x1e\xe6\x00\x18\x00\x18')
Wed Feb 24 20:56:27 2016 CET +0100: skip non-id: 73663ce600e60018e6001800 (b'sf<\xe6\x00\xe6\x00\x18\xe6\x00\x18\x00')
Wed Feb 24 20:56:27 2016 CET +0100: Timeout waiting for printer on port /dev/ttyUSB0; giving up.
Wed Feb 24 20:56:57 2016 CET +0100: Flashing firmware: avrdude -q -q -p atmega1284p -b 57600 -c stk500v1 -P /dev/ttyUSB0 -U flash:w:/usr/lib/franklin/firmware/atmega1284p.hex:i
```
It produce `Upload done : null`.

wijnen, 2016-02-24:
When running it as a service, it uses files from /usr/lib/franklin/.  When running from the commandline, it uses those as a fallback, but files in the current directory will override them.  So if they produce different results, it means the installation of the package (the final step in "make") failed.  This usually happens when building the packages failed.  This is probably because I added some new build dependencies that you don't have installed: doxygen, doxypy and graphviz.

The output looks like it should, but the message should be different.  What happens when you run that last line in the commandline.  Better yet, do so without making it quiet:
``avrdude -p atmega1284p -b 57600 -c stk500v1 -P /dev/ttyUSB0 -U flash:w:/usr/lib/franklin/firmware/atmega1284p.hex:i``

wijnen, 2016-02-25:
The problem when running it as a service was a bug in the code, so sorry for blaming it on you. ;-)  I've fixed it now, so please check if it helped.  The upload should still not report "null" though.  For me it says "firmware uploaded successfully".

clemsept, 2016-02-25:
The result of manual execution of avrdude is the following :
```
clement@debian:~/franklin/server$ sudo avrdude -p atmega1284p -b 57600 -c stk500v1 -P /dev/ttyUSB0 -U flash:w:/usr/lib/franklin/firmware/atmega1284p.hex:i
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x09
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xbd
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x09
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xbe
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0xbf
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x81
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x27
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x27
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x93
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x4b

avrdude done.  Thank you.
```
I have removed and clean franklin. Service is working perfectly and the pop-up during flashing is now functional.
The problem is the same as above (with avrdude), the log is the same. The problem surely come of the board and/or the driver.

We are making progress.
Thank's for your help.

wijnen, 2016-02-28:
That error usually means one of two things.  First, the microcontroller doesn't reset when you make a new connection; in that case, programming never works, except when you press the reset button just before trying.  On the Melzi, you can set this behavior with a jumper.  If there is a jumper shorting the auto-reset pins, this is not the problem.

The other option is that something else is trying to use the serial port.  That can be anything, but a good candidate is the package modemmanager; uninstall it if it was installed.  If it wasn't, you may want to check your list of processes (ps auxfww) to see if there is anything that might have the port open.  lsof may also help to find it.

clemsept, 2016-02-28:
I believe that I found the problem !
I've installed Repetier Server and the melzi is detected with 2 COM ports :
- /dev/tty.USB0
- /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AI031SZ4-if00-port0
When I tried to connect printer with /dev/tty.USB0, that failed. But with the other port : that work !
I will try `avrdude` with this port...

The COM option put me on the right track ! Thank's

wijnen, 2016-02-28:
If you have Repetier Server installed, that is almost certainly the program that is causing your problems; it will try to detect printers on the ports, meaning it writes data to the serial line.  That interferes with avrdude.  Try uninstalling or disabling repetier server; that will likely solve this problem.

clemsept, 2016-02-28:
I installed repetier only 3 days ago and the service must be started manually.
When I'm working with Franklin, I stop the Repetier service.

wijnen, 2016-02-28:
Hmm, ok.  The thing is that the file in /dev/serial/by-id/ is probably a symlink to the other one, so I wouldn't expect it to make a difference.  What does ls -l say for each of them?

clemsept, 2016-02-28:
Ok ! You're right !
It's a link to `ttyUB0` :
```
clement@debian:/dev/serial/by-id$ ls -l
total 0
lrwxrwxrwx 1 root root 13 févr. 28 21:34 usb-FTDI_FT232R_USB_UART_AI031SZ4-if00-port0 -> ../../ttyUSB0
```

clemsept, 2016-03-12:
The problem is not new (from 2014) and it's not resolved : [http://stackoverflow.com/](http://stackoverflow.com/questions/24009955/cannot-program-atmega1284p-with-arduino-on-linux-works-on-mac)

wijnen, 2016-03-12:
Hmm, ok.  The first thing I normally do to a Melzi is replace the bootloader with Optiboot, because it is much better. (In particular, the standard bootloader will wait for 1 second to see if there is a programming attempt, and if incorrect data is received, it will wait for 10 seconds instead.  In contrast, Optiboot correctly assumes it is not being programmed when incorrect data for programming is received, and immediately starts the main program.)

So if you have a way to program the board with ICSP (a second Arduino can do that, although I have heard that this is also not working well on Linux), I recommend installing Optiboot on it.

You can program it directly with avrdude (from Linux or if that doesn't work, the Mac).  You need to connect the wires from the Arduino to the ICSP header, following the information at the top of the ArduinoISP sketch.  The pin names on the Melzi are marked; the table shows which pin on the Arduino to connect them to.  If you have female-female connectors, you can also connect all pins from ICSP to ICSP instead, except for the reset; the reset on the Melzi must go to the correct pin (10 or 53, depending on your Arduino model).

To program the bootloader on the Melzi with this setup, first upload ArduinoISP from the examples to the Arduino.  Then change the board to "Mighty 1284p using Optiboot" and select "Arduino as ISP" as a programmer, then click "Burn Bootloader".

Once you have Optiboot, everything should work.  Of course in Franklin (and in Arduino when using that), you need to select the option with Optiboot.

clemsept, 2016-03-13:
Nice, I'll get right on it!

clemsept, 2016-03-13:
From 3:30pm to 11:00pm, I've tried to flash Melzi with "Migthy-1284p using Optiboot" (and other bootloaders).
Finally, it works : [flash log](https://drive.google.com/open?id=0B0LfnkJrfaWMa2E2X2YyWGlmRzQ)
The hex file was flashed successfully.

But the problem persists with Franklin :'(.
The board is not detected.

wijnen, 2016-03-14:
That's a long time, but it's good that it finally worked.

That file is the bootloader (optiboot) for the microcontroller on the Melzi (atmega1284p).  So after that is flashed, the board doesn't have Franklin's firmware on it yet, but the button to flash the Franklin firmware with Optiboot should work (and after that the detection should also work).

clemsept, 2016-03-14:
Can the Franklin's Firmware be compiled and uploaded manually ? Or with Arduino IDE ?

wijnen, 2016-03-14:
Franklin's firmware cannot be flashed with the IDE, but it can manually.  The file is in /usr/lib/franklin/firmware/atmega1284p.hex.

Clicking the upload button just runs avrdude on it though, so there should be no difference between doing it manually and through the web interface.  But if even with Optiboot you can only get flashing to work on a non-Linux system, it would be a workaround.  I've never had any problems with Optiboot though, so that is strange.

clemsept, 2016-03-14:
Why ?!!
I can upload the firmware manually with these commands :
`sudo avrdude -C/usr/share/arduino/hardware/tools/avrdude.conf -v -v -v -v -patmega1284p -cstk500v1 -P/dev/ttyUSB0 -b19200 -e -Ulock:w:0x3F:m -Uefuse:w:0xFD:m -Uhfuse:w:0xDE:m -Ulfuse:w:0xF7:m
`
AND
`sudo avrdude -C/usr/share/arduino/hardware/tools/avrdude.conf -v -v -v -v -patmega1284p -cstk500v1 -P/dev/ttyUSB0 -b19200 -Uflash:w:/usr/lib/franklin/firmware/atmega1284p.hex:i -Ulock:w:0x0F:m
`

But Franklin doesn't detect the board and the same message appears (not in sync).
In your opinion, is there a problem with the commands ?

wijnen, 2016-03-15:
If you usb the ICSP connector to write anything to flash (-Uflash:...), there is no bootloader anymore (unless that's what you're writing, of course).  So after that, writing new firmware over USB doesn't work anymore, that is normal.

However, when you have uploaded Franklin's firmware, you should not need to upload it again from the web interface and the board should be detected at boot.

This is all very strange.  If you want, can we set up a real time link to go over this?  I would like to understand what is happening.  I'm in the US, so your evening works best for me.

clemsept, 2016-03-15:
Ok. We will do that. What is the best day for you ?

wijnen, 2016-03-18:
This weekend should work.  How about Saturday, 18:00 CET (that's your local time)?

clemsept, 2016-03-18:
Yes, perfect.

wijnen, 2016-03-19:
Please come to https://wijnen.dtdns.net/.  I'm using a self-signed certificate, so expect and override the warning that it gives you.

wijnen, 2016-03-20:
Looking at your log file, I think I understand the problem.  The USB connection is extremely bad.  Franklin is designed to handle bad connections, but it does that by resending the packets when they don't arrive properly.  This works when transmission errors are rare.  From your file, it seems that you are lucky if 10 consecutive bytes make it through.  Franklin is spending all its time trying to receive a complete packet.  This also explains why uploading a new program does not work (other than though ICSP, which does not require accurate timing).

I think the problem may be the operating frequency.  If you can get access to an oscilloscope, try uploading this program:
`void setup() {
  Serial1.begin(100000);
}
void loop() {
  Serial1.write(0x55);
}`

Attach the probes to ground and the tx1 pin on the expansion header.  0x55 is a bit pattern of 1010101010 (start bit, data lsb first, stop bit).  Measure the width of one bit, t.  It should be 1/100000s = 10μs.  If it isn't, that is the problem.  Otherwise, the cable is probably very bad; you probably should try a different cable first anyway, because it is easy.

clemsept, 2016-03-20:
I tried with an other usb cable. But it's the same problem.
Tomorrow, I will analyze bit sequence with an oscilloscope.

clemsept, 2016-03-21:
The width of one bit is correctly 10μs, with a small margin of error (9.9μs measured).

wijnen, 2016-03-24:
That's not a small errror.  It's 10%, a byte is sent as 10 bits (including start and stop); that means the 10th bit is sent at the time that the 9th bit was supposed to get sent.

Here's something you can try: set the fuse bits so that it uses the internal RC circuit as its clock source.  That should give you an 8MHz clock, so when the AVR thinks it sets the baudrate to 115200, it is actually 57600.  Replace those numbers in server.py (not that it occurs more than once) and try again.

The timing will be off (it moves half the speed that you request).  If it works, we can discuss how to fix that.

clemsept, 2016-03-24:
10% are you sure ? I calculated 1% - > (10-9.9)*100/10 = 1.
I will take a look at your solutions.

Once thing is sure, I have communication problems :
When I use Simplify3D (and Repetier firmware on Melzi) the connection sequence start with a lot of special characters. Subsequently, the connection is established.

wijnen, 2016-03-24:
Yes, it's only 1%, you're right.  Sorry about that.  That should not cause any problems.

Repetier firmware has its own protocol, with G-Code as a fallback.  So I think the special characters are its attempt to use it's own protocol; that is probably not an error.

Can you repeat the test while setting the baudrate to 115200?  I wouldn't know why, but perhaps it makes a difference.

wijnen, 2016-04-03:
Did you make any progress?

Did you also have any problems with the Melzi under Windows, or only with GNU/Linux?  If it's just GNU/Linux, the problem is very likely what I wrote before. (You didn't reply, so I'm not sure if you checked?)

> The other option is that something else is trying to use the serial port. That can be anything, but a good candidate is the package modemmanager; uninstall it if it was installed. If it wasn't, you [...]

clemsept, 2016-04-06:
No I'm sorry. I didn't have a lot of time these last weeks.
Moreover, I work on it at workplace with my colleagues (temporarly).
I'll continue my investigations.

To answer your questions, I don't have problems on Windows : the Melzi is detected by several programs and they are able to communicate.
It's only with Franklin (on Debian OS) that I had problems. Furthermore, Repetier Server on Debian detect Melzi (flashed with a Repetier Firmware).

wijnen, 2016-04-07:
But flashing from Debian does not work, correct?  And can Repetier Server on Debian stay connected for a while?  Can it control the printer?  Are there any messages in the console about problems with the serial connection?

This is all from a virtual machine?  Would it be possible to boot a computer into Debian, to make sure the host OS doesn't interfere?  The computer doesn't have to be strong; we've tested it on Beaglebones and Raspberry Pis, both of which work fine.

clemsept, 2016-06-14:
Hey !
I'm sorry for this long delay (I'm not dead, but overbooked).
Due to many complications with the melzi board, I will change my printer board for a RAMPS.
In this case, I could use the melzi for franklin and, besides that, continue printing.

In answer to your questions :
1. Flashing only works manually (With ``avrdude``).
2. Repetier server stays connected with the board when it is flashed with repetier firmware.
But controls don't works properly.
3. I've tried on 3 devices (Raspberry PI, Virtual Debian and Physical Debian). But it is the same issue.

It's possible that the problem comes from the melzi :
I have some problems to upload a recent firmware from Arduino 1.6.X because the available drivers (Hardware files) don't works properly with this mainboard.

wijnen, 2016-06-14:
Hi!

If it works with physical Debian and Raspberry Pi, I'm convinced there is something wrong with the Melzi.  I have no idea what it is, because the detailed tests we did didn't show anything strange, but I'm pretty sure the problem is not in Franklin then.

Using a RAMPS sounds like a good idea; that should work fine, we've been using those on several printers as well.  The only downside is that the auto-reset cannot be disabled, so it cannot recover from a temporarily lost connection. If your printer is in an electronically noisy environment (such as next to a welder), this may be a problem. However, the RAMPS is much better with noisy connections than the Melzi, so it is unlikely that the noise is too much. (In fact, we are using RAMPS boards on our metal printers, where they are next to welders and they work fine.)

Since this doesn't seem to be an issue in Franklin, and I've implemented the interface changes I wanted, I'll close this issue now.  If I missed anything, feel free to reopen it.  If there's a new problem, please open a new issue.

Thanks!

closed 2016-06-14 by wijnen