mtu-most/franklin #11: Probe map in g-code

wijnen, 2016-04-01:
It would be good if g-code could specify its own probe map.  That way, drilling can be done first, then electroplating, then probing and then milling.  The probing will in that case need to avoid the drilled holes.

sheepslinky, 2017-12-22:
This could be added to parsing of G29.  A few firmwares and slicers now handle this via arguments following G29. Drill files could be parsed into keep-out areas and passed through G29.

wijnen, 2018-01-03:
Currently probe maps are done with a rectangular grid.  This cannot continue to be the case if there are keep-out areas.  So first the probing system needs to be changed, probably into something based on triangles.  The maps can then be passed with G29.  However, reading the reprap wiki I don't see any other firmware supporting sending a mesh of points, so it seems some Franklin-specific values would still be required.

Did I miss something, or was this what you meant?

wijnen, 2019-05-28:
Now that the probe is done independently of the print job, it would make sense to use a separate g-code file with just the probe map. This probably doesn't change anything for how this should be implemented.