mtu-most/franklin #19: No options appear after firmware uploaded

wpaulson, 2016-06-04:
As a favor to someone, I'm trying to get a newly assembled Athena printer up and running for the first time.
Using a direct ethernet cable, Franklin's server side at 192.168.76.2:8000/admin is running. 
[image]

The detect button has no obvious results. If I select /dev/ttyO0 and the "Melzi from Beaglebone", it uploads firmware to the Melzi. After upload completes apparently successfully, no options appear, only a blank green screen.
I don't know what version of Franklin is running. It was probably pre-installed on the SD card for the BeagleBoard controller, and I'm not sure when the kit was assembled.

wpaulson, 2016-06-04:
Here's log info from running franklin from the command line:
[output.txt](https://github.com/mtu-most/franklin/files/299012/output.txt)

wijnen, 2016-06-05:
Thanks for the detailed report.

That looks like a very old version of Franklin.  Please install the new version using the latest image from here: http://most.mse.mtu.edu/franklin/
See the readme on that site for instructions on how to install it.  From the log I can see that this Athena has a bridge board version 1; that means you need to get the upgrade from https://github.com/phidiasllc/athena for it, as described in the aforementioned readme.

You need a μSD card (of at least 1GB) for the first step, and a USB thumbdrive for the second.

Please let me know if that solves the problem.  If the installation instructions are unclear, I welcome comments about those as well.

Thanks,
Bas

wpaulson, 2016-06-06:
Thanks for the help. The install succeeded after several tries, with a full Franklin web page now appearing. Now I’ve got to figure why it still doesn’t move.

 A couple of notes:
- Hardest thing was locating the disk images on the net. Without your mail, I may never have found them.
- In the readme file, there’s no instruction for Mac. I suggest “On Mac OSX or Linux, copying the .img file file to the SD card can be done using the dd command from the command line.
	df -h # this will list the disk devices - find the one corresponding to your SD card
	sudo dd if=<.img file> of=/dev/diskXXX bs=64k
It takes about 5 minutes.”
-  After flashing, it doesn’t power down, although it does power down after the update bridge version 1 script runs.

wijnen, 2016-06-07:
Thanks for the feedback. I had no Mac users to ask what worked there, so I didn't know what to write. Thanks for providing the instructions, I'll add them to the document.

If you give some more details, I may help you figure out why it doesn't move.  It would be helpful if you send the profile for that.

wijnen, 2016-06-14:
I've added the instructions to the readme, and a link to the download location to the wiki.  I'll close this issue now; if there are more problems, please feel free to open a new issue.

closed 2016-06-14 by wijnen