mtu-most/franklin #23: Franklin Freezing / connection issues

nlanter95, 2016-09-10:
I believe this issue may have something to do with franklin:

I was printing with no issues and then the printer began to struggle to raise the temperature so I shutdown my printer. When I went to reboot I started having connection issues. Sometimes my beaglebone is just losing power, other times I get a message from my browser saying the connection was lost. If/when I do have a connection I'll try homing the printer or setting the temperature.

If I try to home the printer It'll move a little bit, hit a limit switch, and then stop. Then the uvw grid on the bottom left disappears and after a few seconds, I lose the connection and have to refresh. If I try to put in a new temperature nothing will happen. In each case Franklin will freeze. This is in addition to the connection issues I am experiencing.

Note: I have rebooted numerous times. I am getting a heartbeat and other lights from both the Beaglebone as well as the Melzi board. I am getting lights on the ethernet port on the Beaglebone, and I have tried another patch cable. I am having difficulties with both google chrome and firefox. Also, when I do have a connection the x,y,z are all NaN, it shows which limits are tripped, and the temperature appears to be reading normally.

closed 2016-09-14 by nlanter95

wijnen, 2016-09-15:
Has this been resolved?  What was the problem?

nlanter95, 2016-09-15:
This has definitely not been solved. I did have issues with my Melzi board
which Jerry replaced, but I am still having this issue and it's very
frustrating.

I think I posted about a power supply shutting off. Basically, sometimes
everything will work but then it usually starts giving me problems in this
order:

Can no longer reach target temperature, next time I power up I cannot
change the temperature or home without what I thought was freezing.

The freezing appears to be the power supply cutting off power. Everything
shuts down (sometimes boots back up, sometimes stays off) and the power
supply's LED starts flashing until I unplug it from the printer.

I think it's either a bad PSU, a short, a bad beaglebone or a bad bridge. I
emailed Jerry about meeting to have him look at it and I'm awaiting a
reply.

wijnen, 2016-09-15:
The LED on the power supply brick should never flash.  If that happens, my first guess would be that the mains cable is not fully plugged into it.  If it is, the power supply is faulty.

nlanter95, 2016-09-15:
Yes the cable should be plugged all the way in. The strange thing is that
sometimes it will work. I think the probably bad psu may be linked to my
faulted Melzi board I had replaced.