mtu-most/franklin #24: use of Franklin on Raspberry pi

JeroenVanStappen, 2016-10-27:
Dear Bas,

I am trying to set up the Franklin server on a raspberry pi. As suggested in the Wiki, I have included the Address=0.0.0.0 in the etc/default/franklin file; as well as in the /etc/init.d/franklin file. However, I am now wondering if I still need to connect to the localhost to access the web page. It seems that this connection does not work.

The "make" process for which I sent you a mail previously is now running, with the new dependencies installed (doxygen, doxypy, graphviz), so normally this should not pose any more trouble.

Kind regards,
Jeroen.

JeroenVanStappen, 2016-10-27:
Dear Bas, 

I have made the server running. Instead of using the "Address=0.0.0.0", I have used my eth0 address directly in the server.py script. I now have a different problem, namely that I cannot seem to find my "printer".

As said, I am using a raspberry pi as operating system; and my printer is an Arduino mega 2560, in combination with a Ramps 1.4 reprap board. I have currently only connected one A4988 driver and 1 nema 17 stepper motor, just to try connect these. The stepper motor and driver is connected to the virtual x-axis. 

The problem is that in the franklin server, you have to state the location of the printer ("/dev/ttyAMA0; /dev/ttyS0; or /dev/ttyACM0"). However, the arduino is not connected to any of these locations. the arduino is connected to "/dev/ttyACM1". However, this cannot be chosen. Is there a way to work around this?
Also I tried to follow the steps on the appropedia site of Franklin. When I go to the server website, I cannot click on @/dev/ttyO0.. Maybe this has something to do with it?

I very much appreciate your help,
Kind regards,
Jeroen.

wijnen, 2016-10-27:
Hi,

I changed the setup so it uses the apache webserver.  This means a few things:

* The recommended way of connecting is through http://localhost/franklin/.  (Add "admin" for the admin interface.)
* It is still available at port 8000 as well, but encryption must not be used (so http://, not https://).  If you want encryption, you should set apache up to do that.

If you want to see messages about what's wrong, try this:

sudo service franklin stop
sudo su franklin3d -s /bin/sh -c "/usr/bin/franklin --system --address 0.0.0.0"

This will run the server the same as normally, but in the commandline where you started it, you will see any error and warning (and debugging) messages.

Note that you shouldn't edit /etc/init.d/franklin, because that leads to annoying messages when I change things in it.  So only add the values to /etc/default/franklin.  Also, use all-caps (ADDRESS=0.0.0.0).

Thanks,
Bas

wijnen, 2016-10-28:
Hi Jeroen,

On Thu, Oct 27, 2016 at 08:24:36AM -0700, JeroenVanStappen wrote:
> As said, I am using a raspberry pi as operating system; and my printer is an
> Arduino mega 2560, in combination with a Ramps 1.4 reprap board. I have
> currently only connected one A4988 driver and 1 [...]

That should be fine; the Arduino cannot detect if there is a controller and
will happily send signals to motors that don't exist.

> The problem is that in the franklin server, you have to state the location of
> the printer ("/dev/ttyAMA0; /dev/ttyS0; or /dev/ttyACM0"). However, the
> arduino is not connected to any of these [...]

Are you sure it's not on /dev/ttyACM0?  Do you have another Arduino connected
to the Pi?

In any case, it would be useful if you run the server from the commandline, so
you can see the output; if /dev/ttyACM1 exists but is not in the list, there
will be a message about why it was removed.  Only ports that give an error will
not show up.  For example, the user may not have the rights to access the
device file.

> Also I tried to follow the steps on the appropedia site of Franklin. When I
> go to the server website, I cannot click on @/dev/ttyO0.. Maybe this has
> something to do with it?

No, that's just old; I recently changed the way that works (but I haven't
released that in a stable image yet, so the documentation is not updated).
There used to be one tab per port in the interface; now there is one tab per
printer, which can be connected on any port.  So that's normal; you should just
click on the "+" tab to create a new printer, then select the port and hardware
and upload the firmware.  But of course the port needs to be in the list for
that to work.  So you need to find out why it isn't.

Thanks,
Bas

wijnen, 2017-10-25:
This has been open for a while without response and reading back I'm not sure if there even is a problem still. So I'm closing this issue. If there is still a problem, please open a new issue for it.
Thanks,
Bas

closed 2017-10-25 by wijnen