mtu-most/franklin #27: Under Extrusion on round objects in latest image

lmbrueck, 2017-01-15:
There seems to be an issue that causes very noticeable under extrusion on round objects or parts of objects. I have tried printing the same g-code with the same setup on both the newest image and the "latest stable" image and while the part comes out great with the stable version there is lots of under extrusion on the newest version.

nlanter95, 2017-01-19:
i'd like to add that i also experienced this as well as the extruder zipping around some round corners very quickly, extruding little to no filament. This was very easy to see on the first layer

wijnen, 2017-02-28:
I think the problem here is the arc detection.  Franklin can try to detect that a series of points is really an arc, and move over the arc instead of the points.  The latest stable image didn't support this yet.  However, it seems to be broken, so in the more recently released image, it is disabled again.

wijnen, 2017-12-29:
I've moved the G-Code parsing into the cdriver and plan to add G64 support (rounded corners) to it instead of arc detection. Because there is already an issue for this, and I have worked around the problem that is reported here, I'm closing this issue.

closed 2017-12-29 by wijnen