mtu-most/franklin #28: Make error on a raspberry pi

neilwang0913, 2017-02-28:
Hi Bas:

I cannot wait to try with the latest version control interface with raspberry pi.
After following the instructions in the video on the wiki website, I face an error when I run "Make" command.

The error is logged and please the attachment.
[error.txt](https://github.com/mtu-most/franklin/files/805387/error.txt)

Is there any suggestion to fix this error?

Looking forward your reply.

Kind regards.

Fengzhou

wijnen, 2017-02-28:
Hi Fengzhou,

The end of the message shows the problem:

dpkg-checkbuilddeps: Unmet build dependencies: doxygen doxypy graphviz

You need to install those packages for the build to work.  You can do that using:

sudo apt-get install doxygen doxypy graphviz

(Or use su, if you don't have sudo set up.)

Thanks,
Bas

neilwang0913, 2017-03-01:
Hi Bas:

The server is working with your help.

However, I face some problem when I connect the raspberry pi with a mega 2560 controller(RAMPS  1.3).(see figure 0)
[image]
The webpage shows that: "The machine is not connected".(see Figure 1)
[image]

I setup each configuration and it give the error message on the webpage(see Figure 2) and the command message is shown(Figure 3)

[image]
[image]

Is there any suggestions for fix this problem?

Looking forward your reply.

Kind regards.

Fengzhou

Old55, 2017-04-19:
Similar issues for me. I am trying to connect to a printer with MKS board, which is an ATMega2560. dmesg shows it on ttyUSB0, but running server from terminal shows that it times out when searching for it. Quite frustrating.

Old55, 2017-04-19:
I tried a different ramps 1.4 board (connected to a separate ATMega). When run from the terminal I can see that there is some communication (firmware settings: name, etc.), but it continues to show timeout in the terminal and 'machine not connected' in the browser.

wijnen, 2017-05-29:
Hi.  There seem to be problems uploading the firmware to the atmega2560, regardless of whether it is a Raspberry Pi.  Manually uploading to them seems to work though.  So if anyone can let me know what commandlines are used by Franklin, by "make upload TARGET=atmega2560" and by any other method that works, that would be helpful.  For seeing the command that Franklin uses, it needs to not run as a system service (sudo service franklin stop), but from the commandline (cd server; ./server.py)

sheepslinky, 2017-10-13:
The command issued by franklin is: 
`avrdude -q -q -p atmega2560 -b 115200 -c wiring -P /dev/ttyACM0 -U flash:w:/usr/lib/franklin/firmware/atmega2560.hex:i`

I believe the issue is caused by the avr toolset for arm; at one point I found a bug report on this with avrdude and debian 8 jessie. I have had success programming the at2560 from avrdude 6.3, which is available for debian 9 stretch. 

#1 So, on Raspberry pi, maybe try a distribution upgrade to raspbian stretch. 

#2 Use a workaround. Copy the atmega2560.hex from /usr/lib/franklin/firmware/atmega2560.hex to a computer which has the Arduino IDE. You can then upload the .hex file using the avrdude command that arduino uses. To find this command, upload any sketch to the atmega2560 ("blink" works). In the bottom window, you should see the commands being issued -- you'll see the invocation of avrdude. Copy the command and then change blink.hex to franklin's hex file.  

I'll try playing with some code and inserting newer avr-dude a little and see what happens. It would be nice to get this fixed within franklin.

wijnen, 2017-10-25:
Apparently the bootloaders that do programming over USB do not behave the same as ICP programmers and require -D to be passed on the commandline. This has been done by #34 so this should be fixed now. If the problem still occurs, please reopen this issue.

Arthurivy, 2020-10-22:
I think I have a similar problem now. I am using a MKS Base v1.4 and I try to upload the "Arduino Mega 2560 or Mega ADK (atmega2560, wiring, 115200baud)" firmwate through the browser. However I get the following issue:
[image]

Do you know what could be the problem?

wijnen, 2020-10-22:
I don't know exactly, but I keep having trouble with boot loaders. It may help to hold the reset button on the arduino while you press the "upload" button, then release it and wait for it to complete.

What has been more reliable for me, is to go to the firmware directory in the source, make sure Franklin is stopped and then run:
`make TARGET=atmega2560`

If people have a good idea about what different boot loaders require and how I can make sure it all works (preferably without waiting 10 seconds every time I connect to a device), I'd love to hear it.

Let me know if either of those works for you.

wijnen, 2023-05-31:
I have replaced the Arduino libraries with my own ( https://github.com/wijnen/amat ). This means that this issue is no longer relevant for the current code. There may still be problems with uploads, of course. Please file those as new issues when they come up.

As for the atmega2560, that is not currently supported by amat. Please file an issue about that there if this is still a problem for you. Or better yet, write the mcu description file so it is supported. :-)

closed 2023-05-31 by wijnen