mtu-most/franklin #29: Heated Bed option doesn't work

lmbrueck, 2017-03-09:
In Franklin the Heated bed option doesn't seem to work on either the stable build (6-04-2016) or the latest build (2-16-2017). In the latest build you are unable to even check the box which says bed next to the temp settings. In all cases franklin treats the second temperature as a second extruder.

wijnen, 2017-03-09:
Yes, extruders are not affected by this setting.  It only controls which
temperature is changed in response to M140 and M190 commands.

I'll have a look at the problem.

Thanks,
Bas

lmbrueck, 2017-03-09:
Awesome. thanks Bas.
I would think that checking the "Bed' option should cause that temperature to be controlled by M140 and M190 commands and not be affected by any M104 and M109 commands.

wijnen, 2017-05-29:
They are not coupled.  The bed checkbox means that this temperature is controlled with M140 and M190.  The extruders are always the first temperatures.  If you set one of the extruders to be the bed, it will be controlled by both methods.

The inability to check the box has been fixed, so I'm closing this issue.

closed 2017-05-29 by wijnen