mtu-most/franklin #30: avr dude 6.1-2 on Raspberry Pi (jesse) cannot upload to ATMEGA 2560

sheepslinky, 2017-04-23:
Franklin's command, via AVRDude returns 

`stk500v2_getsync(): timeout communicating with programmer
`
I could not upload the .hex file to the arduino mega via Franklin. Instead, I copied the .hex file to my laptop using avrdude 6.3 (opensuse tumbleweed OS) and uploaded via the command:

`~/arduino-1.8.0/hardware/tools/avr/bin/avrdude -C~/arduino-1.8.0/hardware/tools/avr/etc/avrdude.conf -v -patmega2560 -cwiring -P/dev/ttyACM0 -b115200 -D -Uflash:w:~/atmega2560.hex:i

My workaround worked fine.
`
This seems to be a bug with avrdude and effects the mega2560 and uno mostly.

wijnen, 2017-05-29:
When trying this in a terminal from Franklin, is the command that it uses different?  If you are in the firmware directory, can you flash it with "make upload TARGET=atmega2560"?  Is the command that it uses then different from any of the other two methods?

wijnen, 2017-05-29:
This is the same issue as #28, so I'm closing this.  Please let me know what you find in that issue instead.

closed 2017-05-29 by wijnen