mtu-most/franklin #31: Homing error endstops trigger but motor keeps going

sheepslinky, 2017-04-24:
Bas, 

When home is pressed the motor moves the carriage to the end-stop. It hits the endstop and the franklin UI shows that the endstop was hit (a check mark shows up in V). However, motor does not stop turning and keeps pressing till it stalls.

Franklin in the terminal shows this error:

`cdriver debug from firmware: duplicate 3 2 len: 3
#Error: Setting position while motors are not busy!
#Error: Setting position while motors are not busy!
`  

The error appears when the end stop is hit.

I am running a RAMPS 1.4 board with A4988 stepsticks. 

Thanks

Lorin

wijnen, 2017-05-29:
I think this was a hardware problem and it has been fixed by now, correct?  In that case I will close this bug.

wijnen, 2017-07-04:
Closing this due to lack of reply.  If the issue still exists, please reopen it.

closed 2017-07-04 by wijnen