mtu-most/franklin #37: Tool Head Freezes at finsih of print when Parshing GCode

hsojhcoom, 2018-02-14:
When a print finishes and something its parsing the tool head freezes at the target end of last gcode in the file. It isnt a huge deal but it does make a small dent in the plastic.

wijnen, 2018-02-14:
This problem actually became worse with the new code to parse G-Code with the c driver.  Before, it would freeze if it finished during parsing.  Now it freezes if it's printing while it's parsing.  So it shouldn't allow parsing of G-Code during a print.

On the plus side, parsing because a lot faster, so it's quite possible that it's done before the firmware buffer runs out, in which case there is no problem.

I'm keeping this open because it seems like a good idea to make it impossible to upload new files during a print.

lmbrueck, 2018-02-15:
This doesn't seem to be a problem at all in much older versions were even very long G-Code parsing does not seem to stop the print.

Also would it be possible to either get a new image file posted for Athena or get a bit more guidance on creating these image files? I'd love to try out many of the new features and fixes you are working on but am not sure how to create the image file or update my current one.

wijnen, 2018-11-29:
The last commit fixes this. The parsing is completely rewritten and now it cuts corners (using the max deviation setting or G64 code) so it can move faster. During this change I turned the cdriver into a Python module for better integration. While the cdriver is still a separate process (forked from the module startup code), I put the parsing code into the parent process. This restored the situation where the job keeps running while code is parsing (but the interface is still frozen).

closed 2018-11-29 by wijnen