mtu-most/franklin #45: Temp controls keep driving pins after they are no longer owning them.

wijnen, 2019-06-04:
Copied from chat:
I've been noticing something that I thought was just me, but it seems to keep happening.
If I run into a problem with, say a heater on temp, I disable the pin in the temp, set it up as a gpio and then manipulate the gpio while monitoring output, it frequently doesn't work. It seems like the pin remains associated with the heater pin on the temp. Even if I remove the temp (set number of temps to zero) it still acts like it's associated with the temp.
If I shut he whole thing down while having it set as the gpio and not a heater, and restart (it has to be powered off) then it works as a gpio.
Actually, it doesn't have to be powered off, I can simply restart the OPi, and it will work.

gcanzalo, 2019-06-04:
I'm not sure it's really an issue. Should check if what I've noticed is a phantom or an issue...

wijnen, 2020-06-22:
I believe this has been fixed in the broken branch. I implemented PWM and fixed some issues while I did that. This was one of them.

closed 2020-06-22 by wijnen