mtu-most/franklin #49: Franklin hangs when starting job

gcanzalo, 2019-06-07:
Fri Jun  7 08:07:49 2019 EDT -0400: sending machine '74c5dedf-7a50-cb44-a7b2-f116c924c2fc'
$77:Invalid command i.
$77:Invalid command �.
$77:Invalid command �.
Fri Jun  7 08:08:12 2019 EDT -0400: running /home/debian/.cache/spool/franklin/74c5dedf-7a50-cb44-a7b2-f116c924c2fc/gcode/extruder_axel_x36.bin 0.000000 1.000000

wijnen, 2019-06-07:
Does this happen for any job, or just sometimes? If not always, when does it happen?
What hardware is this on? Is it possible that some other program is trying to use the serial port?

gcanzalo, 2019-06-07:
Occurs when multiple M104 commands are called. A single M104 works fine, more causes failure with no error messages.

wijnen, 2019-06-23:
This error also randomly happens when starting a job with just a single M109 at the start. This means it hangs Franklin regularly, making it quite severe.

commit 2e4754ac (2019-06-23): Don't read temperature from hw in interrupt. Closes #49
closed 2019-06-23 by wijnen