mtu-most/franklin #61: installation error

gcanzalo, 2019-06-26:
Setting up franklin (0.2-1) ...
dpkg: error processing package franklin (--install):
 subprocess installed post-installation script returned error exit status 1
dpkg: dependency problems prevent configuration of franklin-dbgsym:
 franklin-dbgsym depends on franklin (= 0.2-1); however:
  Package franklin is not configured yet.

dpkg: error processing package franklin-dbgsym (--install):
 dependency problems - leaving unconfigured
Processing triggers for systemd (232-25+deb9u11) ...
Processing triggers for mime-support (3.60) ...
Processing triggers for man-db (2.7.6.1-2) ...
Errors were encountered while processing:
 franklin
 franklin-dbgsym

wijnen, 2019-06-26:
This must be because the required Apache modules are not enabled. I have no idea how to do that. In any case, to work around it you should install them manually:

sudo a2enmod proxy proxy_html proxy_http proxy_wstunnel

wijnen, 2019-06-26:
To be clear: what I wrote above is how you do that; I have no idea how to properly do it as part of the package installation. The documentation says I should not enable them, but just set them as dependencies. That's what I do and it's the reason it fails to install: those dependencies are not satisfied. If anyone knows how to fix the package so that it installs the modules on installation, please let me know.

KyleMiles, 2020-04-26:
This should be mentioned in the installation instructions in the readme, seeing as this is a required step ATM.

wijnen, 2020-06-22:
I still haven't found out how this should be done, so for now I've forced the apache modules to be enabled from the Makefile.

closed 2020-06-22 by wijnen