mtu-most/franklin #62: Print failed; subsequent errors on starting from cli

gcanzalo, 2019-06-27:
Franklin froze during a print with feedrate @200%. On starting Franklin from cli:

Wed Jun 26 19:58:20 2019 EDT -0400: starting machine 7cc15208-b127-be41-c492-77c5892655d4
Wed Jun 26 19:58:20 2019 EDT -0400: starting machine f668f738-48ed-ab45-158c-3eaf0d6e258b
Wed Jun 26 19:58:20 2019 EDT -0400: starting machine 9ea1abf3-ba68-9d4f-3c81-a2a4d598b197
Wed Jun 26 19:58:21 2019 EDT -0400: detecting machine on /dev/ttyS1
Wed Jun 26 19:58:21 2019 EDT -0400: Franklin server is running
Wed Jun 26 19:58:22 2019 EDT -0400: accepting known machine on port /dev/ttyS1 (uuid 9ea1abf3-ba68-9d4f-3c81-a2a4d598b197)
Wed Jun 26 19:58:22 2019 EDT -0400: connecting 9ea1abf3-ba68-9d4f-3c81-a2a4d598b197 to port /dev/ttyS1
#Starting
#Starting
Wed Jun 26 19:58:23 2019 EDT -0400: loading profile /home/debian/.local/share/franklin/7cc15208-b127-be41-c492-77c5892655d4/profiles/default.ini
Wed Jun 26 19:58:23 2019 EDT -0400: loading profile /home/debian/.local/share/franklin/f668f738-48ed-ab45-158c-3eaf0d6e258b/profiles/default.ini
Wed Jun 26 19:58:23 2019 EDT -0400: new uuid:'7cc15208-b127-be41-c492-77c5892655d4'
Wed Jun 26 19:58:23 2019 EDT -0400: sending machine '7cc15208-b127-be41-c492-77c5892655d4'
Wed Jun 26 19:58:23 2019 EDT -0400: new uuid:'f668f738-48ed-ab45-158c-3eaf0d6e258b'
Wed Jun 26 19:58:23 2019 EDT -0400: sending machine 'f668f738-48ed-ab45-158c-3eaf0d6e258b'
#Starting
Wed Jun 26 19:58:23 2019 EDT -0400: loading profile /var/lib/franklin/9ea1abf3-ba68-9d4f-3c81-a2a4d598b197/profiles/default.ini
Wed Jun 26 19:58:23 2019 EDT -0400: new uuid:'9ea1abf3-ba68-9d4f-3c81-a2a4d598b197'
#unexpected underrun?
$failed to receive reply to 1c from child: Success (0)
Wed Jun 26 19:58:25 2019 EDT -0400: 9ea1abf3-ba68-9d4f-3c81-a2a4d598b197 died because there was an error.
Wed Jun 26 19:58:27 2019 EDT -0400: sending machine 'f668f738-48ed-ab45-158c-3eaf0d6e258b'
Wed Jun 26 19:58:27 2019 EDT -0400: sending machine '7cc15208-b127-be41-c492-77c5892655d4'
Wed Jun 26 19:58:41 2019 EDT -0400: f668f738-48ed-ab45-158c-3eaf0d6e258b dying as requested by host (Machine is removed).
Wed Jun 26 19:58:41 2019 EDT -0400: exiting as requested by server (2)
Wed Jun 26 19:58:41 2019 EDT -0400: f668f738-48ed-ab45-158c-3eaf0d6e258b died because it was removed.
#cannot read from server: Success (0, 0)
Wed Jun 26 19:58:55 2019 EDT -0400: detecting machine on /dev/ttyS1
Wed Jun 26 19:58:55 2019 EDT -0400: not disabling nonexistent machine 9ea1abf3-ba68-9d4f-3c81-a2a4d598b197
Wed Jun 26 19:58:55 2019 EDT -0400: BUG: port is not in detectable state. Please report this.
Wed Jun 26 19:58:57 2019 EDT -0400: detecting machine on /dev/ttyS1
Wed Jun 26 19:58:57 2019 EDT -0400: not disabling nonexistent machine 9ea1abf3-ba68-9d4f-3c81-a2a4d598b197
Wed Jun 26 19:58:57 2019 EDT -0400: BUG: port is not in detectable state. Please report this.
Wed Jun 26 19:58:58 2019 EDT -0400: detecting machine on /dev/ttyS1
Wed Jun 26 19:58:58 2019 EDT -0400: not disabling nonexistent machine 9ea1abf3-ba68-9d4f-3c81-a2a4d598b197
Wed Jun 26 19:58:58 2019 EDT -0400: BUG: port is not in detectable state. Please report this.

gcanzalo, 2019-06-27:
Retried the print while running from cli. Failed again with no errors:

#a+ 1 0 target v 11.855406 limit dv 8.000000 last v 3.855085 s 1 current 3583.751359 factor 0.000000
#a+ 1 0 target v 27.049254 limit dv 8.000000 last v 6.291178 s 1 current 3583.894429 factor 0.000000
#a+ 1 0 target v 19.704888 limit dv 8.000000 last v 6.291178 s 1 current 3583.894429 factor 0.000000
#a+ 0 1 target v 40.738212 limit dv 12.000000 last v 4.670410 s 1 current -5.367361 factor 0.000000
#a+ 0 2 target v 19.698457 limit dv 12.000000 last v 2.250491 s 1 current 16.618213 factor 0.000000
#a+ 0 1 target v 16.683845 limit dv 12.000000 last v 4.670410 s 1 current -5.367361 factor 0.000000
#a+ 0 1 target v 38.603625 limit dv 12.000000 last v 16.670410 s 1 current -5.300626 factor 0.093996
#a+ 0 1 target v 28.679367 limit dv 12.000000 last v 16.670410 s 1 current -5.300626 factor 0.093996

gcanzalo, 2019-06-27:
Or the host crashed. How to check? Unable to connect via ssh.

gcanzalo, 2019-06-27:
The instant a print is started, CPU load and temperature increase. Response of the website is slow as are other web-based apps, such as OPi-monitor:

[image]

[image]

gcanzalo, 2019-06-27:
[image]

gcanzalo, 2019-06-27:
Third attempt at printing this part failed. Franklin unresponsive:

[image]
[image]

wijnen, 2019-06-27:
During parsing, Franklin makes sure that the global limits for v and a are not exceeded. When setting the feedrate above 100%, this is no longer true. The motor limits are still enforced though.

The CPU heating up when a job is running is normal. Most of the movement math is performed on the host, not in the firmware. I would not expect the load to be at 100%, but it can be significant and lead to an increase in temperature.

So the main problem is that it stops. I have no idea so far what that could be. How does it look? The print just stops? The cpu remains active while it is not printing? Is it more active while it hangs compared to while it prints?

gcanzalo, 2019-06-27:
The print just stops; web interface freezes. Seems temperature is maintained as the fan keeps running. Last night, I was unable to log into the OPi via ssh. Not so this morning as top continued to work. Service reported that Franklin was still active and running.

gcanzalo, 2019-06-27:
Oh, and this morning's test was done at 100% feedrate.

gcanzalo, 2019-06-27:
OPi Zero at 80 C on restart!

gcanzalo, 2019-06-27:
Can confirm that the hot end remains hot. Whether it is controlled or not is unknown. Fan remains on, though.

wijnen, 2019-06-27:
Yes, it should remain controlled. That happens in the firmware, so even if the OPi stops responding, it will continue to control the temperature. If you set a timeout, that will also still work and abort everything when it expires.

Did you close this by accident? It doesn't sound like the issue is solved.

gcanzalo, 2019-06-27:
Sorry, closed by accident...

wijnen, 2020-06-22:
I think this has been fixed when I increased the time between computation steps. It will still get warmer when it's printing, because it's busy. But it shouldn't be excessive. Can this be closed, or do you still see a problem?

wijnen, 2023-05-31:
As there hasn't been a reply for a few years, I'm closing this issue. Feel free to reopen if it is still a problem.

closed 2023-05-31 by wijnen