mtu-most/franklin #65: Cloned but make does not work

patpatbut, 2019-09-08:
Hi there

After clone and I run make and it does not work

dpkg-checkbuilddeps -d 'devscripts, git, wget, sudo,  fakeroot`
dpkg-checkbuilddeps: error: Unmet build dependencies: devscripts
make: *** [Makefile:89:  basedeps] Error 1

I am using Raspbian on pi 3b and still learning on how to use it.
Can you show me how can install it?

Thanks
Pat

wijnen, 2019-09-08:
It needs some packages to work. It runs dpkg-checkbuilddeps to check if they are installed, and that is telling you that they are not. So to fix that, you should install them. Because there are several packages which need to be built, you may get this error (with different missing packages) more times. In each case, just install the packages (in this case: "apt-get install devscripts") and run make again.

On the original RPi there is also a problem to run Franklin, not sure if that is fixed on RPi3. If it installs, but fails to run, Add to /etc/default/franklin:
ADDRESS=0.0.0.0
That way you enable only IPv4. This is required, because IPv6 is broken.

Does that solve your problem?

patpatbut, 2019-09-08:
Hi wijnen

I still have problem to runs dpkg-checkbuilddeps 

dpkg-checkbuilddeps: error: cannot read Debian/control: no such file or directory

Do you know what I should do next? 

I managed to install apt-get install devscripts

Thanks
Pat

wijnen, 2019-09-09:
You don't need to run it manually, make runs it for you.

You need to be in the top level directory of Franklin when you run make.

If you can't find the problem, can you copy the command with all its output into your message?

wijnen, 2020-06-22:
Since there has been no reply for a while, I'm closing this issue. If you still have a problem, please reopen or open a new one. Thanks!

closed 2020-06-22 by wijnen