mtu-most/franklin #66: Park Calibration Issue

brhubbar, 2019-09-27:
I've found that if I park my Athena II mid-print, then resume the print, I lose my calibration - the print resumes offset, missing the existing print by a layer or two, and offset in the horizontal plane. Today I was working on my bed leveling, and I came to find that the distance between my nozzle and the bed depends on whether I translate in the y- or z- direction first. Moving in the z-direction first, my nozzle will hit my calibration spacer, but if I center the nozzle (x = y = 0), then drop the nozzle to the bed, I get about a millimeter of clearance. This does not seem to be a mechanical issue - belts are well-tensioned and the issue is very repeatable.

wijnen, 2020-06-22:
I have made several fixes to the movement recently in the broken branch. If you can check that out and see if it fixes the problem, that would be great. If you don't feel comfortable building "broken", you can wait until I merge it into the main branch. That should be soon.

brhubbar, 2020-06-23:
Hi Bas,

I found that the cause for this issue was drive gear slip. I've applied thread locker (blue) to my drive gears and switched out the set screws for small phillips screws that get better grab, also with thread locker blue. This has since resolved the issue.

I am actively using my 3D printer for work right now, so I will wait until you've merged back into main. At that point, I'd be more than happy to install the new version and provide feedback!

wijnen, 2023-05-31:
Since this issue has been resolved, I'll close it.

closed 2023-05-31 by wijnen