mtu-most/franklin #68: Franklin is not recognizing my printer

eweidensee, 2020-06-06:
Hi all,

I want to preface this by saying I'm fairly sure this is a Franklin issues which is why I'm posting here but I could be wrong. If so, my apologies.  

My printer finished up great print last night no issues, I shut it down, then unplugged (without any issues whatsoever).This morning when I powered up my printer, **Franklin is not showing me all the settings and options I am use to seeing (ex: prints in the queue, temp, print head location, and more) and my printer is not responding to any prompts (like homing or location adjustments). Its like the software is not recognizing my printer.** 
For background I'm using the Melzi/Beagle Bone set up. When I disconnect the Melzi board and tired to reach Franklin the webpage did not show up (as expected). With the Melzi board attached I am seeing the following screen (see images below) and my printer is not responding.
**Any thoughts or advice? Is Franklin down? Is there an update I need?**
The reason I feel that this isn't an issue with my Melzi board is that I just received a second brand new Melzi board and I switched them out this morning and am seeing the same issue. 
I am still able to shutdown my printer by using the following link: http://192.168.76.2/cgi-bin/shutdown

The attached images are screen shots of my Franklin screen upon opening and when checking the source set up button. 

[image]

[image]

wijnen, 2020-06-08:
This is a very old version of Franklin. It's probably a good idea to install a newer version. However, you'll need to download and compile that, which is not trivial. (Depending on how you have the network set up, it may be pretty easy or very hard.)

In any case, it looks like your profile is broken for some reason. Do you still have the (working) profile saves somewhere? Importing it probably solves the current problem.

eweidensee, 2020-06-08:
Thank you so much for your reply! Do you have any ideas what would have caused the potential broken profile? I made no known changes and the printer was working fine ~20 hours before.

I'm not very familiar with franklin and do not know where my working profiles would have been saved to. If there are some where on my PC I will look for them.

As for updating, I'm not quite sure how to do so if the following links are not the correct places. I build my printer in fall 2018 with the following instructions and have used this version of Franklin ever since.
Instructions: https://www.appropedia.org/Franklin:MOST
File location: http://most.mse.mtu.edu/franklin/

As for my set up: I am using a windows PC connected to the beagle bone via ethernet cable. The beagle bone is connected to my Melzi board via a small bridge board. 
Are there other instructions I should be looking at? My other thought is to just reinstall the version I've used previously following the instructions linked above.

wijnen, 2020-06-12:
The profiles can be exported to your computer. If you've done this while it was still working, you can now restore it.

If you haven't, there's probably no way to get it back. The profile on the beaglebone is obviously broken, so knowing where it is doesn't help. (But for completeness: it is under /var/lib/franklin/.)

If you're using a setup with a network cable plugged into your computer, the Beaglebone does not have access to the internet, which makes updating pretty hard. I wouldn't try it at the moment.

The easiest way to get it working again is probably to use a micro-SD card to flash the system onto the Beaglebone again. This will reset everything to its original state (so you do need to recalibrate).

I'm not sure how this could have happened. It's possible that the memory on the Beaglebone is bad. In that case you can expect random failures to happen, so hopefully that's not the problem.

Another possibility is that the filesystem filled up somehow. In that case it may have tried to write a new version of the profile, but it failed. It can destroy the existing profile in the process, I think.

wijnen, 2020-06-22:
Since this doesn't appear to be an issue with Franklin, I'm closing this. Feel free to continue replying if you have any problems.

closed 2020-06-22 by wijnen