mtu-most/franklin #71: More then 3 axis

Daniel-Kurth, 2020-07-09:
Hi, I'm planing a 3D printer project based around a delta printer which would require me to control up to 12 steppers via gcode:
- 3 steppers for the delta towers
- 2 Extruders
- 7 steppers additional axises

Is this possible with this firmware?

wijnen, 2020-07-10:
Hi, thanks for your interest in Franklin.

What you want is not currently possible, because some functionality is broken (#57). It probably isn't possible when that's resolved either, but it'll be easy to add. Also, depending you your exact needs, there may be a workaround.

Here's the situation:

Franklin firmware supports as many motors as the AVR that you use has pins and memory for. The main ones that people use are the atmega2560 (used on Ramps) and atmega1284 (used on Melzi), both of which can easily support 10 steppers; 12 should probably work as well.

The firmware has no limits on which motors can move at which time, but the host software does. It splits the steppers into different types:
3 main position axes (x, y, z) are controlled by 3 steppers (x,y,z/u,v,w/r,Θ,z/... depending on the geometry)
3 additional axes (a,b,c) are controlled by 3 steppers (a,b,c). This feature is currently broken.
an unlimited number of extruders are controlled by 1 stepper each.
an unlimited number of followers (which copy the moves of some other stepper, but with their own limit switch and possibly inverted pins) are controlled by 1 stepper each.

So (when a,b,c is working again), assuming your additional steppers cannot be set up as followers, Franklin can do everything you want, except it only supports 3 additional axes, not 7. The reason for this is G-Code: I'm not aware of a notation for specifying more axes. If there is one, it's trivial to implement it.

The upside of using a,b,c is that it makes it possible to move all the axes simultaneously. However, if you don't require this, there is a workaround. If you define the additional axes as extruders, it is possible to control them. The only limitation is that each move can control exactly one extruder alongside the position axes. So if your needs only require the axis to move, not to move all (or some) of them at the same time, defining them as extruders would solve your problem.

Please let me know what you want to use them for. And if you're aware of a way to define simultaneous moves of some (if possible, unlimited)  axes with G-Code, I'd also very much like to know about it. Similarly if there is a notation for moving more than one extruder in a single move.

Thanks again,
Bas

wijnen, 2023-05-31:
This seems to be sufficiently answered, so I'm closing this issue.

closed 2023-05-31 by wijnen