mtu-most/franklin #74: Unable to locate doxypy

oliviayang23, 2022-05-02:
I do not have any experience with softwares/programs. I am using the makefile on Raspbian on raspberry pi. Every time i run make it shows error code
unable to locate package doxypy
unable to locate package | 
unable to locate package python3-doxypy

I have tried to install these packages using sudo apt-get install doxypy...etc, but I still get this error code

wijnen, 2022-05-04:
Hi,

Thanks for the report. I'll fix it soon. To make it work, edit Makefile and on
line 93, change "python3-doxypy" into "python3-doxypypy".

Thanks,
Bas

oliviayang23, 2022-05-04:
Hi Bas, 
Thanks so much for the reply, 
I have changed line 93, but now when I run the make there is another error with dpkg-checkbuilddeps: error: unmet build dependencies: pybuild-plugin-pyproject 
+ exit 1
fatal: run_command returned non-zero status for python-fhs

thanks!

wijnen, 2022-05-12:
The solution to this is to install the things it complains about:

sudo apt install pybuild-plugin-pyproject

I'll add it to the list of tiles that are installed automatically; thanks for letting me know.

wijnen, 2022-05-12:
Hm, in that case it needs a different solution.

I'll do some rewriting. The easiest way around it is by doing:

sudo apt install python3-pip
pip install websocket-httpd
make mkdeb
./mkdeb
sudo dpkg --force-depends --install /tmp/franklin.deb

Note that it should complain about python-fhs, python-network and python-websocketd; those are not a problem. If it complains about anything else, install those packages.

Please let me know if that works.

brhubbar, 2022-05-15:
Ah.. this would have been good to read before I bricked my orange pi trying to update. I had some issues with the motors jittering around (moving farther than requested, crashing, a whole bunch of great stuff) and the thermal controller switching at the incorrect temperature, maintaining the temperature higher than set. Upon trying to reinstall the latest version, I've been fighting these same errors.

That said, your final command fails:

```shell-session
dpkg: error: cannot access archive '/tmp/franklin.deb': No such file or directory
```

This seems to be why:

```shell-session
athena@athena-ii:~/franklin$ ls /tmp/
franklin-0.2                    franklin_0.2-1.dsc                                                       systemd-private-8d094c2197a644deb38b9a02ef5a262e-chrony.service-u6cZOi
franklin_0.2-1_armhf.build      franklin_0.2.orig.tar.gz                                                 systemd-private-8d094c2197a644deb38b9a02ef5a262e-haveged.service-qVpmrg
franklin_0.2-1_armhf.buildinfo  franklin-dbgsym_0.2-1_armhf.deb                                          systemd-private-8d094c2197a644deb38b9a02ef5a262e-systemd-logind.service-RXQEig
franklin_0.2-1_armhf.changes    hsperfdata_root                                                          systemd-private-8d094c2197a644deb38b9a02ef5a262e-vnstat.service-b96c6e
franklin_0.2-1_armhf.deb        ssh-HXfwkr7Lxc
franklin_0.2-1.debian.tar.xz    systemd-private-8d094c2197a644deb38b9a02ef5a262e-apache2.service-r1Jhhg
```

May I ask why you aren't shipping git submodules in the franklin repo (as opposed to pulling the latest during install)? Your work on those other repos broke this installer.

brhubbar, 2022-05-15:
Scratch that - looks like I can call franklin after running one of those two. Rebooting, franklin is not started up automatically.

I was able to resolve this with a cron job:

```shell-session
athena@athena-ii:~$ crontab -e                                   > Add '@reboot /usr/bin/franklin'
no crontab for athena - using an empty one
crontab: installing new crontab
athena@athena-ii:~$ sudo reboot
```

wijnen, 2022-05-16:
I'm not using submodules because I don't like to have multiple copies of things on my disk. Also the idea is that people would just have the packages installed; they aren't supposed to be a part of Franklin. But they weren't released, so installing them is hard. That's why I made the hack in the Makefile to install them.

But I didn't like that, so I have just released all of them. They are now installable with pip (with non-intuitive names, unfortunately; the intuitive ones weren't allowed) and they have also just been accepted into Debian (just unstable for now, but they will migrate automatically). In other words, I'm working on removing that.

The error message you got on install is different. I still need to figure out how to solve it properly. In the Makefile it runs a workaround for it, but I forgot to give it in my instructions. You need to do:

sudo a2enmod proxy proxy_html proxy_http proxy_wstunnel

Installing the deb after that should work. Since you already installed it, you only need to configure it:

sudo dpkg --configure --pending

brhubbar, 2022-05-20:
```diff
athena@orangepizero:~/franklin$ git diff Makefile
diff --git a/Makefile b/Makefile
index befc60b..0be52df 100644
--- a/Makefile
+++ b/Makefile
@@ -90,10 +90,16 @@ clean: $(addprefix clean-,$(MODULES))
 
 rundeps = python3-serial, avrdude, adduser, lsb-base, apache2
 franklindeps = debhelper, python3-all, dh-python, gcc-avr, arduino-mighty-1284p, arduino-mk, python3-all-dev
-depdeps = doxygen, doxypy | python3-doxypy, libjs-jquery, graphviz, openssl
+depdeps = python3-pip, doxygen, python3-doxypypy, libjs-jquery, graphviz, openssl
 basedeps:
        deps=`dpkg-checkbuilddeps 2>&1 -d 'devscripts, git, wget, sudo, fakeroot, ${rundeps}, ${franklindeps}, ${depdeps}'`; deps="$${deps##*:}"; if [ "$${deps}" ]; then echo "Installing $$deps"; sudo apt install $$deps ; fi
        dpkg-checkbuilddeps
 
 .PHONY: install build clean zip bb basedeps
        
+bens-build: basedeps mkdeb
+       pip install websocket-httpd fhs-paths network-wrapper
+       ./mkdeb
+       sudo a2enmod proxy proxy_html proxy_http proxy_wstunnel
+       sudo dpkg -i --force-depends /tmp/franklin*.deb
+
```

Making this change, then calling `make bens-build` worked for me. I see now that `websocket-httpd` includes the other two as requirements. 

Any idea why franklin isn't automatically kicking off on boot? I expected the install to set up a service or something similar to autostart it, instead of requiring me to add a cron job for it.

brhubbar, 2022-05-20:
As it turns out, franklin is not detecting my printer now. Having bricked my orange pi, I pulled the pi off the printer for re-flashing. Now that I've hooked it back up, all I'm getting is this:
[image]

```shell-session
athena@athena-ii:~$ franklin
22:32:22 franklin:<module>:133:	whitelist: 
22:32:22 franklin:detect:748:	detecting machine on /dev/ttyS6
stty: /dev/ttyS6: Input/output error
22:32:22 franklin:detect:777:	failed to open serial port /dev/ttyS6 (Could not configure port: (5, 'Input/output error')).
22:32:22 franklin:detect:748:	detecting machine on /dev/ttyS4
stty: /dev/ttyS4: Input/output error
22:32:22 franklin:detect:777:	failed to open serial port /dev/ttyS4 (Could not configure port: (5, 'Input/output error')).
22:32:22 franklin:detect:748:	detecting machine on /dev/ttyS2
stty: /dev/ttyS2: Input/output error
22:32:22 franklin:detect:777:	failed to open serial port /dev/ttyS2 (Could not configure port: (5, 'Input/output error')).
22:32:22 franklin:detect:748:	detecting machine on /dev/ttyS0
stty: /dev/ttyS0: Permission denied
22:32:22 franklin:detect:777:	failed to open serial port /dev/ttyS0 ([Errno 13] could not open port /dev/ttyS0: [Errno 13] Permission denied: '/dev/ttyS0').
22:32:22 franklin:detect:748:	detecting machine on /dev/ttyS7
stty: /dev/ttyS7: Input/output error
22:32:22 franklin:detect:777:	failed to open serial port /dev/ttyS7 (Could not configure port: (5, 'Input/output error')).
22:32:22 franklin:detect:748:	detecting machine on /dev/ttyS5
stty: /dev/ttyS5: Input/output error
22:32:22 franklin:detect:777:	failed to open serial port /dev/ttyS5 (Could not configure port: (5, 'Input/output error')).
22:32:22 franklin:detect:748:	detecting machine on /dev/ttyS3
stty: /dev/ttyS3: Input/output error
22:32:22 franklin:detect:777:	failed to open serial port /dev/ttyS3 (Could not configure port: (5, 'Input/output error')).
22:32:22 franklin:detect:748:	detecting machine on /dev/ttyS1
stty: /dev/ttyS1: Input/output error
22:32:22 franklin:detect:777:	failed to open serial port /dev/ttyS1 (Could not configure port: (5, 'Input/output error')).
22:32:22 franklin:<module>:976:	Franklin server is running
Franklin server is running
22:32:25 websocketd.py:page:1240:	file builders.js not found in /usr/lib/franklin/html
22:32:25 websocketd.py:page:1240:	file rpc.js not found in /usr/lib/franklin/html
```

wijnen, 2022-05-20:
The original reason that Franklin didn't start was that the package failed to configure. That was fixed by the a2enmod line.
The new reason is that builders.js and rpc.js are installed by the websocketd package, but not by the pip install. I need to fix that. As a workaround you can get them from the source ( https://github.com/wijnen/python-websocketd ) and place them in /usr/share/python3-websocketd/ (which you need to create).

I don't know why it's not finding the machine. Is it plugged into the USB port? In that case there should be a file named /dev/ttyUSB0 or /dev/ttyACM0. Looking at the error message, that does not seem to be the case. This suggests there may be something wrong with the Melzi. Are there messages in "sudo dmesg" about it?

brhubbar, 2022-05-28:
Copying those files in removed that error. Running `sudo franklin` gave it the ability to access `/dev/ttyS0`. 

Franklin is only offering three boards to connect to; the athena II is not an option

[image]

[dmesg.log](https://github.com/mtu-most/franklin/files/8791505/dmesg.log)

wijnen, 2022-06-01:
If you set the port to /dev/ttyS1 (which should be the port that the AthenaII board connects to), there should be an option named "Athena on OrangePi Zero (atmega1284p at 12MHz)". That's the one you need.

brhubbar, 2022-06-02:
Looks like [this](https://forum.odroid.com/viewtopic.php?t=34289) was the missing link. Looking back at my backup of the Pi before I bricked it, I changed the `overlays` line in `/boot/armbianEnv.txt` to match what it had been. Before modifying, `i2c1`, `spi-spidev`, and `uart1` were not present. The missing `uart1` was likely the culprit.

```txt
verbosity=1
bootlogo=false
console=serial
disp_mode=1920x1080p60
overlay_prefix=sun8i-h3
overlays=i2c1 spi-spidev uart1 usbhost2 usbhost3
rootdev=UUID=bb089afa-e0fe-47d8-a3d5-9e704e2adcfa
rootfstype=ext4
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u
```

Franklin has once again found my printer - maybe. After some fighting with Armbian to try to convince it to stop changing my MAC address upon reboot (I lost) franklin crashed upon attempting to home. I'm clocking out for the night, will post what I learn tomorrow.

brhubbar, 2022-06-02:
Still broken. I have a machine showing up, but it's non-responsive. Uploading to ttyS1 caused the motors to move around as I'd expect, but the new machine is not operational.

```shell-session
08:10:37 franklin:<module>:133:	whitelist: 
08:10:37 franklin:detect:748:	detecting machine on /dev/ttyS6
stty: /dev/ttyS6: Input/output error
08:10:37 franklin:detect:777:	failed to open serial port /dev/ttyS6 (Could not configure port: (5, 'Input/output error')).
08:10:37 franklin:detect:748:	detecting machine on /dev/ttyS4
stty: /dev/ttyS4: Input/output error
08:10:37 franklin:detect:777:	failed to open serial port /dev/ttyS4 (Could not configure port: (5, 'Input/output error')).
08:10:37 franklin:detect:748:	detecting machine on /dev/ttyS2
stty: /dev/ttyS2: Input/output error
08:10:37 franklin:detect:777:	failed to open serial port /dev/ttyS2 (Could not configure port: (5, 'Input/output error')).
08:10:37 franklin:detect:748:	detecting machine on /dev/ttyS0
stty: /dev/ttyS0: Permission denied
08:10:37 franklin:detect:777:	failed to open serial port /dev/ttyS0 ([Errno 13] could not open port /dev/ttyS0: [Errno 13] Permission denied: '/dev/ttyS0').
08:10:37 franklin:detect:748:	detecting machine on /dev/ttyS7
stty: /dev/ttyS7: Input/output error
08:10:38 franklin:detect:777:	failed to open serial port /dev/ttyS7 (Could not configure port: (5, 'Input/output error')).
08:10:38 franklin:detect:748:	detecting machine on /dev/ttyS5
stty: /dev/ttyS5: Input/output error
08:10:38 franklin:detect:777:	failed to open serial port /dev/ttyS5 (Could not configure port: (5, 'Input/output error')).
08:10:38 franklin:detect:748:	detecting machine on /dev/ttyS3
stty: /dev/ttyS3: Input/output error
08:10:38 franklin:detect:777:	failed to open serial port /dev/ttyS3 (Could not configure port: (5, 'Input/output error')).
08:10:38 franklin:detect:748:	detecting machine on /dev/ttyS1
08:10:38 franklin:<module>:976:	Franklin server is running
Franklin server is running
08:10:39 franklin:boot_machine_input:879:	accepting unknown machine on port /dev/ttyS1
#Starting
08:10:40 driver.py:user_load:1872:	not loading nonexistent profile
08:10:40 franklin:get_vars:523:	new uuid:'9fa51031-c371-bd4a-dab1-cd2e4784669e'
08:10:40 franklin:finish:886:	finish detect '9fa51031-c371-bd4a-dab1-cd2e4784669e'
08:10:40 franklin:finish:889:	connecting new machine 9fa51031-c371-bd4a-dab1-cd2e4784669e to port /dev/ttyS1
#Finished connecting
New machine 9fa51031-c371-bd4a-dab1-cd2e4784669e connected on port /dev/ttyS1
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
08:10:58 driver.py:_close:485:	disconnecting
08:10:58 franklin:upload:310:	Flashing firmware: sudo /usr/lib/franklin/opi/flash-opi /usr/lib/franklin/opi/avrdude.conf /usr/lib/franklin/firmware/atmega1284p-12MHz.hex
[sudo] password for athena: 
[... 91 lines cut ...]
#Finished connecting
Machine 9fa51031-c371-bd4a-dab1-cd2e4784669e connected on port /dev/ttyS1
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
#Too much silence; request packet to be sure
```

wijnen, 2022-06-02:
It looks like it doesn't have a profile loaded, but otherwise it seems fine. Do you have a profile saved? Loading one is a lot easier than entering all the pins manually.

I've fixed some errors yesterday and pushed them to the broken branch. If you're not using that, you should: one of the problems could cause it to crash while homing.

I'm planning to make it a release soon.

brhubbar, 2022-06-03:
Oof. Forgetful me.

Uploading my previous profile kinda worked - I got an error that the park order wasn't right, and then homing didn't seem to be working. Instead of troubleshooting I went about building the latest from broken like you suggested. 

Just built the latest from broken (had to comment out the franklindeps that I manually installed because they haven't made it to stable yet). I appreciate the cleanup on the Makefile, it's a lot easier to track what's happening now. The install failed at the final step - for some reason, `$(wildcard` didn't work as expected:

```shell-session
dpkg-shlibdeps: warning: debian/franklin/usr/lib/franklin/franklin-cdriver contains an unresolvable reference to symbol __aeabi_atexit@CXXABI_ARM_1.3.3: it's probably a plugin
   dh_installdeb
   dh_gencontrol
   dh_md5sums
   dh_builddeb
dpkg-deb: building package 'franklin-dbgsym' in '../franklin-dbgsym_0.2-2_armhf.deb'.
dpkg-deb: building package 'franklin' in '../franklin_0.2-2_armhf.deb'.
 dpkg-genbuildinfo
 dpkg-genchanges  >../franklin_0.2-2_armhf.changes
dpkg-genchanges: info: including full source code in upload
 dpkg-source --after-build .
dpkg-buildpackage: info: full upload (original source is included)
+ test '' -a '' -a ''
+ exit 0
sudo dpkg -i 
dpkg: error: --install needs at least one package archive file argument

Type dpkg --help for help about installing and deinstalling packages [*];
Use 'apt' or 'aptitude' for user-friendly package management;
Type dpkg -Dhelp for a list of dpkg debug flag values;
Type dpkg --force-help for a list of forcing options;
Type dpkg-deb --help for help about manipulating *.deb files;

Options marked [*] produce a lot of output - pipe it through 'less' or 'more' !
make: *** [Makefile:46: package] Error 2
athena@athena-ii:~/franklin$ /tmp/franklin-
franklin-0.2/      franklin-jlg7wyaw/ franklin-x3oc__09/ 
```

Upon manually installing (and using force depends to bypass the lack of python3-websocketd and company):

```shell-session
athena@athena-ii:~/franklin$ sudo dpkg -i --force-depends /tmp/franklin*.deb
(Reading database ... 51188 files and directories currently installed.)
Preparing to unpack /tmp/franklin_0.2-2_armhf.deb ...
Unpacking franklin (0.2-2) over (0.2-2) ...
Preparing to unpack .../franklin-dbgsym_0.2-2_armhf.deb ...
Unpacking franklin-dbgsym (0.2-2) over (0.2-2) ...
dpkg: franklin: dependency problems, but configuring anyway as you requested:
 franklin depends on python3-websocketd; however:
  Package python3-websocketd is not installed.
 franklin depends on python3-network; however:
  Package python3-network is not installed.
 franklin depends on python3-fhs; however:
  Package python3-fhs is not installed.

Setting up franklin (0.2-2) ...
Installing new version of config file /etc/default/franklin ...
Installing new version of config file /etc/init.d/franklin ...
Adding user _franklin to group dialout
Created symlink /etc/systemd/system/multi-user.target.wants/franklin.service → /lib/systemd/system/franklin.service.
apache2_invoke franklin: already enabled
Setting up franklin-dbgsym (0.2-2) ...
Processing triggers for mailcap (3.69) ...
Processing triggers for man-db (2.9.4-2) ...
athena@athena-ii:~/franklin$ 
```

Franklin kicked up and detected my device. However, upon trying to `detect` or `upload`, I get this error:

```python 
error: Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/websocketd.py", line 644, in _recv
[... 47 lines cut ...]
Jun 02 21:54:24 athena-ii start-franklin[3679]: stty: /dev/ttyS3: Input/output error
Jun 02 21:54:24 athena-ii start-franklin[3579]: 21:54:24 franklin:detect:777:        failed to open serial port /dev/ttyS3 (Could not configure port: (5, 'Input/output error')).
Jun 02 21:54:24 athena-ii start-franklin[3579]: 21:54:24 franklin:detect:748:        detecting machine on /dev/ttyS1
Jun 02 21:54:24 athena-ii start-franklin[3579]: 21:54:24 franklin:<module>:976:        Franklin server is running
Jun 02 21:54:25 athena-ii start-franklin[3579]: 21:54:25 franklin:boot_machine_input:879:        accepting unknown machine on port /dev/ttyS1
Jun 02 21:54:26 athena-ii start-franklin[3682]: Traceback (most recent call last):
Jun 02 21:54:26 athena-ii start-franklin[3682]:   File "/usr/lib/franklin/driver.py", line 53, in <module>
Jun 02 21:54:26 athena-ii start-franklin[3682]:     import numpy
Jun 02 21:54:26 athena-ii start-franklin[3682]: ModuleNotFoundError: No module named 'numpy'
Jun 02 21:54:26 athena-ii start-franklin[3579]: 21:54:26 franklin:die:576:        None died because there was an error.
Jun 02 21:54:26 athena-ii start-franklin[3579]: 21:54:26 franklin:get_vars:520:        failed to get vars
Jun 02 21:54:48 athena-ii start-franklin[3579]: 21:54:48 franklin:_call:402:        No active machine
Jun 02 21:54:57 athena-ii start-franklin[3579]: Traceback (most recent call last):
Jun 02 21:54:57 athena-ii start-franklin[3579]:   File "/usr/local/lib/python3.9/dist-packages/websocketd.py", line 644, in _recv
Jun 02 21:54:57 athena-ii start-franklin[3579]:     self._call(data[1][0], data[1][1], data[1][2], data[1][3])
Jun 02 21:54:57 athena-ii start-franklin[3579]:   File "/usr/local/lib/python3.9/dist-packages/websocketd.py", line 662, in _call
Jun 02 21:54:57 athena-ii start-franklin[3579]:     call((lambda ret: self._send('return', (reply, ret))) if reply is not None else None, getattr(self._target, member), *a, **ka)
[... cut ...]
Jun 02 21:54:57 athena-ii start-franklin[3579]:     wake(wake)
Jun 02 21:54:57 athena-ii start-franklin[3579]:   File "/usr/local/lib/python3.9/dist-packages/websocketd.py", line 453, in wake
Jun 02 21:54:57 athena-ii start-franklin[3579]:     return ret.send(arg)
Jun 02 21:54:57 athena-ii start-franklin[3579]:   File "/usr/bin/franklin", line 303, in upload
```

I noticed the missing `numpy` module. Upon a `pip install numpy`, I'm having trouble finding enough room in tmpfs to complete the action, I'll work on troubleshooting that tomorrow.

wijnen, 2022-06-03:
To flash the firmware it uses sudo, but you have it set up to require a password for that. It obviously doesn't handle that well. I'm not sure about the best way to solve that, but I do know how you can avoid the problem: allow the  _franklin user to use sudo without a password. To do this, I have this line in /etc/sudoers (edit that file using "sudo visudo" to make sure you don't end up with a broken system):

%sudo   ALL=(ALL:ALL) NOPASSWD:ALL

And make sure the _franklin user is in group sudo, using

adduser _franklin sudo

It's possible that this used to be set up correctly; I renamed the user from "franklin3d" to "_franklin", because that's the new Debian policy. So if you see "franklin3d" mentioned anywhere on your system, you should replace it with "_franklin".

As for numpy: thanks for letting me know, I forgot to add the dependency. The easiest way to install it, which doesn't require a large /tmp, is through the Debian package: sudo apt install python3-numpy

wijnen, 2022-06-03:
The solution is pretty simple: the package should add permissions for using sudo without a password for _franklin.

I've just done that, and pushed the result ti github (in broken). So if you build the new package, it should work. You shouldn't need to upload again, so the printer should work fine without it, but if you can test that uploading works without changing the sudoers file, that would be great.

brhubbar, 2022-06-03:
That did the trick. Got this error upon importing my profile from last year:

```shell-session
Parse errors:
invalid value for park_order (park_order = 0.000000)
invalid value for park_order (park_order = 0.000000)
invalid value for park_order (park_order = 0.000000)
```

Upon trying to home, one of my steppers gets real upset and starts jerking around, not really moving toward where it is supposed to. This is what originally started me on this re-install journey. Now I'm wondering if I have a bad stepper or driver (though sometimes it moves just fine)

The other thing I'm noticing (before and now) is that the temperature controller is rising to ~20*C over what it is set to. The fan turns on when I expect, but the steady state temperature is too high. I didn't notice this the first time around, but I'm wondering if that has been the case since the latest release. I never went through to calibrate the PID, so it's still just using the default proportional control (Infty, 0, 0)

[image]
[image]

wijnen, 2022-06-03:
The motor seems to have a hardware issue indeed. In my experience, the motor itself is unlike to be bad. The driver could be. But I would first look if all the wires are connected securely. A half-loose connection in a screw terminal (most likely) or inside a wire could cause this. It's also important to fix, because it may lead to damage to the driver.

For the temperature I'm not sure why it would do that, but I would suggest to put finite values in. I have the PID set to 3, 10, 0. It has more overshoot than it should, but it eventually stabilizes to the value that is set.

You can ignore the errors about the park order. Those used to be (incorrectly) stored as floating point values. I have changed that to integers, but that means it does not understand values with a decimal point in them anymore. So those values were not used, and instead it falls back to the defaults. But since the values were all 0, which is also the default, it has no effect on the profile.

brhubbar, 2022-06-03:
Sure enough - loose wire. Shoulda looked under the hood before mentioning that one :)

Still not happy with how those settings are behaving. Even with P=3 and I=0 (right half of image, left half has I=10), it's turning on the heater before the measured temp is less than the set temp. 

Re: overshoot, it's been a while since I designed one of these, but a non-linear cheat that I've used in the past is not kicking the integral controller in until the temp is within 10% error of the set temperature, and/or setting a saturation limit on the integral control so it doesn't get so wound up. If you pointed me to where the code handles that control, I could try to implement that in the next few weeks.

[image]

wijnen, 2022-06-04:
It does look like it turns on the heater too soon and I don't know why. Maybe I added a hack at some point that I forgot to remove. It's also worth noting that the graph has fewer points than are actually measured, so it is possible that it reaches a lower point than what is shown. It may also be smoothing the result, which means that the noise may get below the set value when the heater is turned on. But I still wouldn't expect this kind of response from that, so probably something else is happening.

If you want to look at it, that'd be great. The PID handling is in the last part (starting at line 241) of server/cdriver/temp.cpp .

brhubbar, 2022-06-04:
Couple thoughts looking through for the first time:

- `server/cdriver/packet.cpp`, line 457: `//Add 10% if hold time == 0 and PID is set up` - looks like that's the cause for the high temperature (10% is pretty big when operating in Kelvin)
- In the PID, division by `I` means that the I control is always going to be saturated at 2 if `I=0`. I'd just have users provide Kp, Ki, and Kd directly instead of `Ki = Kp/Ti` and `Kd = Kp*Td` (so Kp is only in part_P, Kd is only in part_D, etc)
- Looks like if P, I, and D are set to 0, the PID is ignored. Haven't found where that happens yet.
- Why are there 3 copies of `arch/` in the repo?
- What is the purpose of `hold_time`, and what controller takes over when that's non-zero?

[image]

[image]

- KI displays on the UI like it's an int, even though it is a float behind the scenes. Kp = 10 and Ki = 0.1 yields a pretty smooth response.

[image]

I think just removing that PID scaling hack in `packet.cpp` would make the PID operate really well. If anti-windup is still of interest, I can do that in a few weeks (I'm leaving town tomorrow). Here's an portion of a PI thermal controller I built last year:

excerpt from thermalcontroller.h:

```cpp
  /*
   * PI controller parameters. ctrl_p is a value proportional (by KP) to the
   * error between the set and measured temeperatures. ctrl_i is a bounded
   * value that cumulatively accounts for the error (scaled by KI), providing a
   * static control input when the system has no error.
   *
   * When integral controllers are implemented digitally, they must have a
   * constant sample rate (CTRL_PERIOD). This known rate is used to scale each
   * addition to the cumulative input ctrl_i so that the dynamics are
   * predictable. (Adding 1 to a value every 10 ms vs every 100 ms results in a
   * very different response from the controller).
   *
   * When a system is very slow, or the input is limited (as in this case),
   * the integrator can experience 'windup', where its contribution to the
   * control input gets very large, so that the max input to the system is
   * provided for much longer than it should, making the system take a long
   * time to settle down. To prevent this, the integrator is not activated
   * until the system is close to its set temperature, and it's bounded to
   * limits that allow it to stabilize the system, but not go overboard.
   *
   * The two control values are summed together in u. U is a value in Watts
   * representing the amount of power the heaters should provide to the system.
   * Since the heaters are either on (300 W) or off (0 W), the heaters must
   * switch on and off at a set rate to provide an average power in equal to u.
   *
   * The ratio of u to the max power input, U_SAT, is the percentage of time
   * that the heaters should be on. The relay switches on and off over a set
   * period of time, DUTY_PERIOD. This period was selected to be slow enough to
   * allow for good resolution on distinct dutycycles (101 possible values)
   * while being fast enough for the system to behave as though it were
   * receving a constant power input.
   *
   * The dutycyle is updated by the controller every CTRL_PERIOD milliseconds.
   * This does not mean that the relay has to update at that rate. Instead,
   * the relay is physically limited to update every ~1.5 cycles of its AC
   * load, or 26-31ms for 50-60 Hz mains power. Since the relay can change
   * state every RELAY_PERIOD milliseconds, the relay can only represent
   * floor(DUTY_PERIOD/RELAY_PERIOD)+1 distinct duty cycles.
   *
   * Given the above constraints, some memory-saving measures can be taken.
   */
  const float KP = 10.;
  const float KI = 0.1;
  float error;
  float ctrlI;
  float ctrlP;
  float u;
  uint8_t dutycycle;
  uint8_t relayState;
[... 109 lines cut ...]
  }

  static unsigned long cycleStart = currentMillis;
  unsigned long dutyTime = currentMillis - cycleStart;
  if (dutyTime > DUTY_PERIOD) {
    // If the cycle period has ended, mark the start of a new period.
    cycleStart += DUTY_PERIOD;
    // Update dutyTime to reflect this change.
    dutyTime = currentMillis - cycleStart;
  }

  if (dutyTime < DUTY_PERIOD * (dutycycle/100.)) {
    // We're in the first part of the dutycycle, so the relay should be on.
    relayState = HIGH;
    return PASS;
  }

  // We're in the latter part of the dutycycle, so the relay defaults to off.
  // If the controller is moving so slow that the whole period is passing
  // before the state can update, the relay will stay off (safer than staying
  // on).
  relayState = LOW;
  return PASS;
}
```

wijnen, 2022-06-05:
Haven't looked at your code yet, but here's my response to your observations:

- server/cdriver/packet.cpp, line 457: //Add 10% if hold time == 0 and PID is set up - looks like that's the cause for the high temperature (10% is pretty big when operating in Kelvin)

Yes, this is certainly the reason for the higher value. However, I now remember that this is needed because of this:

- What is the purpose of hold_time, and what controller takes over when that's non-zero?

It's intended for systems that have a (physical) relay controlling the heater. Some heated beds have that. I don't want to PWM them to death. So it will just refuse to change the value of the output for a while after it has been changed.

The system that takes over is "enable if the temperature is too low, disable otherwise". This is implemented in the firmware and it ensures that the machine won't be set on fire when the host disconnects from the microcontroller.

The 10% is indeed pretty big, but that is intentional: the PID may need to enable the heater when the current temperature is quite a bit higher than the target value. The value that is set (when PID is active) is therefore just a fire protection setting, not something that is meant to actually be reached.

- In the PID, division by I means that the I control is always going to be saturated at 2 if I=0.

Yes, this is correct. The unit is seconds; it is the time in which the I buffer should reach its correct value. (With the unrealistic assumption that nothing changes during that time, so this isn't actually the time in seconds, but the order of magnitude is correct.) To disable the I, it should be set to Infinity.

- I'd just have users provide Kp, Ki, and Kd directly instead of Ki = Kp/Ti and Kd = Kp*Td (so Kp is only in part_P, Kd is only in part_D, etc)

No, I like it better this way. Both are common, but I prefer this because the units are simple. For example, I is in s, otherwise it would be in s*K/% or something incomprehensible like that). Because of this, users can have some intuition for what a sensible value is.

- Looks like if P, I, and D are set to 0, the PID is ignored. Haven't found where that happens yet.

I don't think this is the case. However, with I set to 0, its influence will always override everything so effectively it disables the PID.

- Why are there 3 copies of arch/ in the repo?

The firmware and host software both need arch-specific code. So each of them has its own directory. But I would like to make it easy to port the system to a new architecture, so I want a single place where all arch-specific files are. This is similar to the type-system where I have moved all files that define the machine type (cartesian, delta, etc) into one place.

The directories from the firmware and host directories are just links to the "real" place where the data is stored.

wijnen, 2023-05-31:
This seems to be resolved, so I'm closing this issue. If it is still a problem, please reopen or file a new issue.

closed 2023-05-31 by wijnen