mtu-most/franklin #75: Unable to communicate with RAMPS 1.4 board using RPi 4

bxpqxd, 2023-02-14:
Hi there! I'm doing a school project based on https://www.appropedia.org/Open-source_syringe_pump. I am new to RPi and RAMPS. While I've gotten franklin to run successfully, it does not seem to be able to communicate with RAMPS. Please provide guidance.

**Background:**

- RPi Serial Port (enabled), Serial Console (disabled)
- Connected to RAMPS 1.4 board using both USB and serial connection over breadboard; also tried disconnecting the USB and vice versa
- Followed tips on https://github.com/mtu-most/franklin/issues/74 to get franklin running
- Updated server.py to bypass AMA0 as that was causing the web page to hang

**During Server Startup:**
- read_boards() detecting "Mighty 1284p using Optiboot" and "Original Mighty 1284p"
- console messages include:
  - detecting machine on /dev/ttyS0
  - detecting machine on /dev/ttyUSB0
  - Franklin server is running
  - Timeout waiting for machine on port /dev/ttyS0; giving up
  - Timeout waiting for machine on port /dev/USB0; giving up

**When navigating to website:**
- The site loads properly
- First dropdown shows /dev/ttyS0 and /dev/ttyUSB0
- Second dropdown shows Mighty and Original Mighty (and additionally Melzi when ttyS0 is selected)

**When clicking on the Upload button:**
- The site shows "uploading firmware for [selected option]"
- console messages include:
  - Flashing firmware: [details]
  - broadcasting to one
  - detecting machine on [selected port]
  - broadcasting to one
  - Timeout waiting for machine on port [selected port]; giving up
- The site shows a popup dialog with the message:
  - upload done: firmware upload for [selected option]:
  - avrdude: stk500_recv(): programmer is not responding
  - avrdude: stk5000_getsync() attempt 1 of 10: not in
  - sync: resp=0x00
  - [continues such messages for another 9 attempts]

wijnen, 2023-05-31:
Thanks for the detailed report.

The problem I see is that the dropdown does not show an option for the atmega2560 (which is the microcontroller on the RAMPS board), so uploading will never work. I'm not sure why that is missing. It may have to do with changes in the Arduino libraries.

A complication is that I recently changed the underlying system away from the Arduino libraries; I now use my own system ( https://github.com/wijnen/amat ). While this is better in general, I haven't written the microcontroller description file for the atmega2560, so that doesn't currently work at all anymore. Also, I haven't updated the upload system in Franklin's interface, so currently the way to upload the firmware to the microcontroller is by running "make upload" in the firmware directory (with some options on how to upload). This new system is all in the "broken" branch; nothing is in stable yet.

I'm hoping to get both of those issues fixed soon. I don't want to work on tracking the changes in the Arduino code, especially because I'm moving away from it. However, if you want to try to get stable to work, my guess would be that you need to either move the boards.txt file from the Arduino code to the place where it is searched for, or better: search for it in the place where it is now.

wijnen, 2023-06-03:
Update: I have now added support for the atmega2560 to amat, and also to Franklin. Plus I have implemented the firmware uploading code again. So things should just work with the broken branch now.

I'm planning to release it as stable soon, so please let me know if there are any problems with it.